ccrisan/motionpie #214: Picture gallery breaks if pictures removed outside of UI?

github-0, 2015-06-09:
Hi,

I removed a few days of motion captured pictures and now the UI shows an error when trying to look at the gallery. Is this expected and should I only remove pictures/videos using UI tools?

The motioneye.log shows about 8000 lines of 
2015-06-09 21:55:26: ERROR: stat failed: [Errno 2] No such file or directory: '/data/output/2015-06-05/16-26-38.jpg'

and then
2015-06-09 21:56:15: ERROR: timeout waiting for the media listing process to finish

ccrisan, 2015-06-10:
1) How did you remove the images? Using the UI? Remotely, using SSH? Using SMB/network share access?
2) When exactly does the error appear? Immediately after opening the media browser? When clicking on a group (day)? When clicking on a picture?
3) The first error can be safely ignored.
4) Check to see if the second error's timestamp corresponds to the time when you got the error. The error indicates that there are (were) many media files to analyze and that the analysis process did not finish in time (currently within 120 seconds).
5) Can you easily reproduce the problem or was it just a one time issue?

github-0, 2015-06-10:
1) First from SMB/network share using Windows explorer (this is when the problem initially started), later I deleted some more through SSH
2) After clicking the camera icon the error appears after two minutes (red text on the UI "Failed to get pictures list.") and then fades away
4) It corresponds. 
5) I can reproduce the problem any time

Hardware is Pi B.

ccrisan, 2015-06-10:
Approximately how many pictures do you have stored on your card? The symptom corresponds 100% to the case where there are way to many pictures and the system simply cannot analyze that amount of pictures within 2 minutes.

github-0, 2015-06-10:
Well damn... I had approximately 9.000 pictures left right now and once I deleted them, the gallery is working again.

I didn't think that was the problem (after deleting a whole bunch of them) because the gallery worked earlier with over 10.000 pictures and by looking at the log it seemed like the timeout was caused by the system looking up already deleted pictures. 

Btw, the reason for having so many pictures is that in my case it's impossible not to get false positives even with a mask (sun + wind = too many moving shadows everywhere). I can look up the pictures outside the UI so this isn't a huge problem, but if you're willing to develop the gallery function further, I'd suggest a time window (date/time) the user can choose so the UI doesn't have to look up so many pictures at once.

Glad it works now and sorry for taking your time.

ccrisan, 2015-06-10:
I agree the media browser needs some enhancements. It was developed for a PC where the disk is fast and the CPU is not a problem. On a Raspberry PI things aren't too happy when it comes to reading from the "disk".