ccrisan/motionpie #216: Wrong Date Time

LiviuShiva, 2015-06-12:
Hi Calin,

It is a very nice project and very useful.
I tested on a Raspberry PI 2.

I noticed that, after reboot, the date became 1970.01.01 00:00:00 + TimeZone.
Even changed it on Linux, the date turns back to 1970.

A small suggestion. Maybe is better to have  a  textbox to confirm the password.

Thank you for sharing the project !

Liviu

ccrisan, 2015-06-13:
Make sure your PI has a working Internet connection. More precisely, it needs to be able to access google's server using HTTP/80. If you think Internet access is not an issue, please send me the boot.log file.

LiviuShiva, 2015-06-13:
Hi,

It is connected on internet:
[image]
[image]
 and I'm sure the port 80 is not closed for it.

The boot looks like:
---- booting ----
Setting hostname: done
Starting logging: done
Starting watchdog: done
Starting eudev: done
Setting up data partition
Setting custom root/admin password
Starting wpa_supplicant: done
Starting network: wlan0: starting dhclient
eth0: starting dhclient
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.21  netmask 255.255.255.0  broadcast 192.168.0.255
        ether b8:27:eb:98:01:8e  txqueuelen 1000  (Ethernet)
        RX packets 3  bytes 702 (702.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2  bytes 700 (700.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        loop  txqueuelen 0  (Local Loopback)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.193  netmask 255.255.255.0  broadcast 192.168.0.255
        ether 00:87:34:6d:55:41  txqueuelen 1000  (Ethernet)
        RX packets 10  bytes 1265 (1.2 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 4  bytes 1012 (1012.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

done
Starting sshd: done
Setting current date using http: failed
Starting http date updater: done
Starting proftpd: done
Starting crond: done
Setting smb admin password: done
Starting smbd: done
Starting nmbd: done
Starting motioneye: done
(I will send you the file via mail)

I changed Date Method to NTP, it takes a little longer, but it's working.

I can try to see I have the same issue on Raspberry PI B+

Thanks you,

Liviu

LiviuShiva, 2015-06-14:
Hi,

I incidentally found the cause of the issue. Under my router I have enabled Norton Connect Safe for DNS-based Filtering. Without it, working perfectly.

Thank you !

ccrisan, 2015-06-15:
It's great to hear you got it working.

closed 2015-06-15 by ccrisan