ccrisan/motionpie #218: Picture gallery: "An error occurred. Refreshing is recommended."

github-0, 2015-06-14:
I'm instantly getting this message when trying to view pictures/videos through the UI. Here is a snippet from MotionPie.log:

2015-06-14 14:36:31: ERROR: [Errno 12] Cannot allocate memory
Traceback (most recent call last):
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1288, in _stack_context_handle_exception
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1475, in wrapper
  File "/programs/motioneye/src/handlers.py", line 704, in get
    self.list(camera_id)
  File "/programs/motioneye/src/handlers.py", line 121, in wrapper
    return func(self, *args, **kwargs)
  File "/programs/motioneye/src/handlers.py", line 798, in list
    callback=on_media_list, prefix=self.get_argument('prefix', None))
  File "/programs/motioneye/src/mediafiles.py", line 318, in list_media
    process.start()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 130, in start
  File "/usr/lib/python2.7/multiprocessing/forking.py", line 121, in __init__
OSError: [Errno 12] Cannot allocate memory
2015-06-14 14:36:31: ERROR: 500 GET /picture/1/list/?_=1434281797220&_username=admin&_signature=censored (192.168.33.1) 80.86ms
2015-06-14 14:36:48: ERROR: Exception in callback <functools.partial object at 0xb5b17c60>
Traceback (most recent call last):
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/ioloop.py", line 565, in _run_callback
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/stack_context.py", line 275, in null_wrapper
  File "/programs/motioneye/src/thumbnailer.py", line 71, in _run_process
    _process.start()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 130, in start
  File "/usr/lib/python2.7/multiprocessing/forking.py", line 121, in __init__
OSError: [Errno 12] Cannot allocate memory
2015-06-14 14:37:48: ERROR: Exception in callback <functools.partial object at 0xb60b8d80>
Traceback (most recent call last):
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/ioloop.py", line 565, in _run_callback
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/stack_context.py", line 275, in null_wrapper
  File "/programs/motioneye/src/thumbnailer.py", line 71, in _run_process
    _process.start()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 130, in start
  File "/usr/lib/python2.7/multiprocessing/forking.py", line 121, in __init__
OSError: [Errno 12] Cannot allocate memory
2015-06-14 14:38:48: ERROR: Exception in callback <functools.partial object at 0xb5b17c60>
Traceback (most recent call last):
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/ioloop.py", line 565, in _run_callback
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/stack_context.py", line 275, in null_wrapper
  File "/programs/motioneye/src/thumbnailer.py", line 71, in _run_process
    _process.start()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 130, in start
  File "/usr/lib/python2.7/multiprocessing/forking.py", line 121, in __init__
OSError: [Errno 12] Cannot allocate memory

The memory allocation messages keep repeating every minute even when the UI/camera isn't accessed. Tried refreshing, different browser, different device, rebooting. I have 200 files in the gallery (SMB share) and they are being saved correctly as we speak.

ccrisan, 2015-06-14:
What device/model do you use? The error literally indicates that the system ran out of memory.

github-0, 2015-06-15:
Pi B+. 

Well it seems like all my problems solve themselves after posting them here. I deliberately waited two days to see if it goes away, but now it started working.. Still have no idea what caused it though.

ccrisan, 2015-06-15:
512 MB of ram may not be enough in some scenarios. The motion daemon also eats up a lot of memory itself.

closed 2015-06-15 by ccrisan