ccrisan/motionpie #220: Camera Not Supported

Cyril25, 2015-06-16:
Hi,
I've a Raspberry Pi 2 with motionpie and I've connected an USB camera with no issue, but for the second one, I see the message "camera not supported", 
I bought this camera on aliexpress http://www.aliexpress.com/snapshot/6506544487.html?orderId=65926957675420
Do you think that this camera is compatible with motionpie or not ?
Do you have a list of “low price & good image quality” ip camera compatible with motionpie ?
Thank you for your help

ccrisan, 2015-06-16:
There's no list of supported IP cameras. Unless the IP camera has support for MJPEG streaming, it won't work with motionPie. I couldn't find this information in the link you sent me. Where do you see the "camera not supported" message? Could you send me a screenshot?

Cyril25, 2015-06-19:
Sorry for the delay, I was not at home,
Here attached the screenshot,

[image]
[image]

I can see the video directly with the web interface of the camera, but it's not working with motionpie
Do you need something else ?
Thank you for your help

ccrisan, 2015-06-19:
The message is a bit different than what you said and indicates that the network camera most likely uses HTTP Digest authentication which is not supported by motion. I'm sorry but there's nothing I can do about this.

closed 2015-06-19 by ccrisan

Cyril25, 2015-06-19:
OK thank you for tour help.Can you suggest me some network camera with good image quality, not very expensive and working with motion please?ThxCyril

-------- Message d'origine --------