ccrisan/motionpie #221: writeimage script on OS X

yohannver, 2015-06-17:
Hi,

thanks for this great distribution. I tested it on my raspberry 2 but today i want to do the same for my A+.
I'm using the writeimage.sh script with mac os X 10.10.3.
The command i executed is : "sudo bash writeimage.sh -d /dev/disk2s1 -i "/Users/yohannver/Raspberry/motionpie-raspberrypi-20150331.img" -n "xxxx:xxxxxx""
The first time i had this error : Missing mount_fuse-ext2 for EXT4 mounting! Further configuration stopped.
See http://osxdaily.com/2014/03/20/mount-ext-linux-file-system-mac/
how to install ext4 mount support, please include 'Enabling EXT Write Support'.

I installed fuse-ext2 but i got new errors :
[image]

Thank you for your help.

Yohann

ccrisan, 2015-06-18:
Use -d /dev/disk2.

yohannver, 2015-06-18:
Great it works using disk2 like you said :)
I think i have another issue. I plugged in the sd card into my A+, it boots and after 5 minutes the pi light and the camera pi light are red but the wifi configuration does not work.
I'm using an edimax usb wifi.

ccrisan, 2015-06-18:
Please send me the logs found on the third partition in the `log` directory.

yohannver, 2015-06-19:
Hi,

i used another sd card and did the same thing and it worked :)

Thanks for your help