ccrisan/motionpie #224: Incorrect clockspeed settings for Pi 2.

lukeroge, 2015-06-20:
All the overclock profiles and clock speeds are not correct for the Pi 2. All of the overclock profiles are still lower then the stock CPU speed on the Pi 2.

Additionally, I should be able to set my own overclock settings. If I set stock settings, your software overwrites them with the "Turbo" settings which cause my Pi to fail to boot.

ccrisan, 2015-06-20:
Overclock presets are the same for both PI1 and PI2 and are taken from [raspi-config](https://github.com/asb/raspi-config/blob/master/raspi-config). The "stock" CPU speed on PI2 is 900 MHz so that would be the *medium* overclock preset. I don't see how the *turbo* preset (1000 MHz) is lower than the stock (900 MHz).

"My software" will overwrite your settings but not always with *turbo* but with whatever you have chosen in the UI. If *turbo* is too much for your PI, choose a lower one. If you need to manually set the various frequencies to other values, then use Raspbian + motionEye.

lukeroge, 2015-06-21:
Sorry, I was a bit mistaken about the default clockspeeds on the Pi 2. Regardless, it would be nice to add the "Pi2" preset from raspi-config as it seems to work best on the Pi 2.

ccrisan, 2015-06-21:
Yes, that would be a good idea indeed.

closed 2015-06-21 by ccrisan