ccrisan/motionpie #226: StreamEye fps drop to .5fps

snagglewest, 2015-06-22:
Raspberry Pi 1 Model B+ & CSI camera
EW7811Un WiFi Adapter (150Mbps 11n)
V 20150621

I use Zoneminder for a front end with MotionPie just streaming, no motion, no saving images etc so StreamEye piqued my interest.  With MotionPie I'm getting an average of 9-10fps according to Zoneminder with these settings:

FPS - 15
Quality 50% (no resizing)
Size - 640X480

With the same settings using StreamEye it starts about 13-14fps, which is a nice improvement, but after 15-20mins fps drops off horribly to an average of .5fps (1/2fps)

I understand this is a pre-release and there's probably some bugs to work out, but is this normal?  Any recommendations for settings to improve performance?

Also, would it be possible to add a FPS meter to the GUI without significantly impacting performance?

Snagg

ccrisan, 2015-06-22:
The fps drop is definitely not normal. Can you test it a bit further? Does it ever come back to normal? Could it be a network issue? Do you run with the debugging option enabled?

snagglewest, 2015-06-22:
I can test further if you'd like just let me know what you want me to try.  As far as network, I don't believe it's network related as MotionPie is rock solid maintaining  an average of 9-10fps and has for months.  I also don't have debugging on but I can enable it if you wish.

ccrisan, 2015-06-22:
Could you ssh into your PI while the framerate is 0.5 fps and see what "top" and "free" have to say? There are two log files that are of special interest in this case: `/var/log/raspimjpeg.log` and `/var/log/streameye.log`. Please send them to me as they are. Then, you could enable debugging but just for *debugging* purposes; don't leave it on afterwards as it will generate huge log files.

snagglewest, 2015-06-22:
Email sent

ccrisan, 2015-06-23:
@snagglewest thanks for the logs. The good news is that everything looks clean and that the CPU and MEM usage are low. The bad news is that I cannot tell why the frame rate is that low.

1) Did you capture the logs right after the frame rate decreased to 0.5fps?
2) When the framerate in zonemonder drops drastically, can you open the mjpeg stream in Firefox and see if the problem is visible there as well?
3) Can you try opening the mjpeg stream in Firefox, let it stream for 20 minutes or so and see if the fps problem appears in this case as well?

snagglewest, 2015-06-26:
Life got in the way for as couple days, but I've had a chance to test some more.  I started with a fresh install and changed out the power supply which helped a bit but I still get a drop to 1.5-2.5fps in zoneminder.  When the fps dropped I opened firefox, and while it's hard to tell exactly how much without a fps meter in the interface, it's clear the fps is low.  I've tested this on both of my RP B+ and get the same results. I've switched both back to regular mode and I'm getting an average of 8-10fps in ZM.

ccrisan, 2015-06-26:
Well there's another scenario that may be worth testing: new http connections every few seconds. What I have tested myself is running streameye + firefox for a few hours. It worked like a charm, firefox recreating the http connection just a couple of times. Zoneminder on the other hand appears to establish a new http connection relatively often. I'll try to recreate that precise scenario.

snagglewest, 2015-06-26:
Any chance of incorporating a fps meter in the gui or in the text overlay so we can see what the fps is while viewing the stream?

ccrisan, 2015-07-06:
After some thorough tests, I discovered what the problem is in your case. I'll try to explain it in a few words.

Reading jpeg data into memory and writing it to a client are two mechanisms that are synchronized. The reason is that we want the jpeg data to be consistent when it gets to the client (browser, viewer, zoneminder, etc).

Now, when a client connection becomes slow for some reason, the entire process is slowed down because of this. This is what I noticed in your debug logs. Zoneminder appears to launch up to 3 clients (I'm not sure why or if they are all active). At a certain moment, writing data to one of the clients started to lag and eventually timed out.

I see the following solutions to this problem:

1. Remove the synchronization between the read and client threads. This will possibly serve invalid jpegs to clients from time to time.
2. Keep the synchronization but use a temporary buffer for the client, to prevent blocking by problematic clients. This is the most "correct" solution, but requires an additional memory copy operation which is CPU intensive, especially at higher resolutions or quality.
3. Leave the system as it is and rely on the client and network not to misbehave (efficient and easy but rarely the case).
4. Add a *max clients* limit, which, if set to 1, will disconnect/discard any existing clients upon a new incoming client. This way we know that Zoneminder will always have one active client connection to motionPie. The downside is that connecting with another client (e.g. a browser) will terminate the Zoneminder's current connection.

I'm open to other ideas.

snagglewest, 2015-07-08:
1. Remove the synchronization between the read and client threads. This will possibly serve invalid jpegs to clients from time to time.

2. Keep the synchronization but use a temporary buffer for the client, to prevent blocking by problematic clients. This is the most "correct" solution, but requires an additional memory copy operation which is CPU intensive, especially at higher resolutions or quality.

4. Add a max clients limit, which, if set to 1, will disconnect/discard any existing clients upon a new incoming client.
-----------------------------------------------------------------
I looked at the fast cam interface and didn't see any of the tweaks you mentioned so I assume that's something you'd have to add or is there a way I can add the tweaks through SSH?  If so where/how?

I'm more than willing to test for you if you want to explore these further.  I agree #3 is not a real option in the real world so that's a non-starter but I think the other three are worth exploring if you're up to it.

ccrisan, 2015-07-08:
The tweaks are still in my mind, nothing has been implemented yet :) I will use the solution number 2 and will try to minimize the overhead as much as possible.

ccrisan, 2015-07-14:
@snagglewest I've been working on a new version of streameye that should normally fix all of the problems while providing the promised performance. Could you please test it before I include it in a release? I just need you to download it from [here](https://www.dropbox.com/s/65j5ftmic8p8j8e/streameye?dl=0) and copy it over the existing /usr/bin/streameye on your motionPie.

If you use a Linux PC, you could simply insert the SD card into your PC and overwrite the file. Otherwise you'd need to enable debugging in Expert Settings and the use SFTP to overwrite the file.

Please keep me posted.

snagglewest, 2015-07-14:
I managed to DL the file but how do I get it on the SD?  I have windows 7, no linux machine with a SD card reader.

I've used pscp in the past but when I try this string it keeps kicking an "unable to open /usr/bin/streameye: failure" error.  Maybe a permissions issue???  I'm a bit linux challenged :)

c:\pscp.exe -v c:\testing\streameye admin@192.168.1.231:/usr/bin/streameye/streameye

------------------------------------------
Looking up host "192.168.1.231"
Connecting to 192.168.1.231 port 22
Server version: SSH-2.0-OpenSSH_6.8
Using SSH protocol version 2
We claim version: SSH-2.0-PuTTY_Release_0.63
Doing Diffie-Hellman group exchange
Doing Diffie-Hellman key exchange with hash SHA-256
Host key fingerprint is:
ssh-rsa 2048 a7:f7:61:7e:d1:f5:7e:a2:db:4a:f1:bf:95:bb:4e:67
Initialised AES-256 SDCTR client->server encryption
Initialised HMAC-SHA-256 client->server MAC algorithm
Initialised AES-256 SDCTR server->client encryption
Initialised HMAC-SHA-256 server->client MAC algorithm
Using username "admin".
Welcome to mp-2513b725!
Attempting keyboard-interactive authentication
Server refused keyboard-interactive authentication
admin@192.168.1.231's password:
Sent password
Access granted
Opening session as main channel
Opened main channel
Started a shell/command
Using SFTP
Connected to 192.168.1.231
Sending file streameye, size=28123
pscp: unable to open /usr/bin/streameye: failure
Sent EOF message
Server sent command exit status 0
Disconnected: All channels closed
------------------------------------------

ccrisan, 2015-07-14:
pscp is fine, but don't add a second /streameye in the destination path; it's just /usr/bin/streameye.

snagglewest, 2015-07-14:
Not sure why it keeps changing the font and size on some of my text... its not intentional.

Anyhow, dropped the second streameye and same issue

Sending file streameye, size=28123
pscp: unable to open /usr/bin/streameye: failure

snagglewest, 2015-07-14:
Ok, got it transferred.  Didn't have debugging on :)

It's working but I'll do some testing and let you know.

snagglewest, 2015-07-16:
The streameye update seems to be working much better!  I haven't noticed any drop off with it running 24 hours now and the fps is improved too.  The average was approx 10-11fps (wireless) before and now it averages about 14-15fps and I've noticed as high as 22fps watching on Zoneminder.

ccrisan, 2015-07-17:
@snagglewest this is great news. Thanks for taking the time to test it. The new stable release will hopefully be released this weekend.

snagglewest, 2015-07-17:
I probably should have mentioned just for comparison purposes that my settings are:
Raspberry Pi B+
standard Wi-Pi 150M adapter
video resolution - 800X600
Image Quality - 25
Overclocking - medium
Everything else stock

Thanks for all your hard work, it's pretty impressive performance you've been able to squeeze out of the little Pi!

Any plans to incorporate a FPS meter in the GUI?

ccrisan, 2015-07-19:
@snagglewest, incorporating a FPS meter is very difficult with the current Fast Network Camera approach. The JPEGS that are prepared directly by the GPU are streamed via a small HTTP server to the browser. There's no (easy) way for me to measure the actual number of frames per second in this scenario and overlay it on the JPEG.

The only possible way to achieve this is to separately make the browser query the FPS from the HTTP server every now and then and display it somewhere next to the camera frame.

Anyway this is definitely not a priority so don't expect it to be implemented any time soon.

closed 2015-07-19 by ccrisan