ccrisan/motionpie #227: 5Ghz Wifi USB Stick

Jasimo, 2015-06-22:
Hi,
I plan to switch from a 2,4Ghz WiFi USB Stick to 5Ghz one because all 2,4Ghz channels are busy here in my location. Is there a way to find out which 5Ghz Stick is suppported by the actual MotionPie (BuildRoot) release? When it works out of the box with raspbian is it then also possible with MotionEye?

ccrisan, 2015-06-22:
Normally all adapters that work with Raspbian should work with motionPie. If you find one that doesn't, I'll be happy to include the driver for it.

closed 2015-06-22 by ccrisan

Jasimo, 2015-07-16:
Just bought a:
TP-Link Archer T4U AC1200 Dualband Wireless LAN Adapter
http://www.tp-link.us/products/details/cat-11_Archer-T4U.html

and give it a try, maybe (if not working out of the box) you can implement the drivers ;-)
rgs
Jan

ccrisan, 2015-07-16:
That must have been expensive :) Well, as soon as Raspbian gets it working, motionPie will follow shortly, unless, of course it already works.

Jasimo, 2015-07-20:
Unfortunately its not workng "out of the box" with Raspbian 3.18.11-v7+ #781.
I compiled the following driver, which takes some research because of missing kernel headers for 3.18.11-V7+, but finally i got it.
[rtl8812AU_8821AU_linux](http://github.com/abperiasamy/rtl8812AU_8821AU_linux)
I load the kernel-module with:
sudo modprobe -a 8812au
Is there a way for me to get this running also in MotionPie by myself or is it better to go with MotionEye?

ccrisan, 2015-07-20:
I have many requests like yours to integrate out-of-tree drivers into motionPie. It's a very unpleasant and tedious task and to be honest I really don't like doing it. Let's hope the kernel 4.0 that Raspbian recently started to use will work with your device. Anyways, until then, you're better off with Raspbian + motionEye. After all, if you managed to compile an out-of-tree kernel driver you probably know your way around Linux.

Jasimo, 2015-07-20:
Will do, keep up the good work.
rgs
Jan