ccrisan/motionpie #229: Pi Camera goes blank and then no stream

EHIRL, 2015-06-23:
Firstly great project, very nice UI.

I'm using a RPI B, the first revision out, WIFI and the RPI CSI camera. Everything seemed fine for the evening but when I went to the webpage in the morning stream was a blank black box, when I rebooted it said it could not get image from the camera or something to that effect. It was the second day in a row that it happened. The 1st time it happened I flashed a new image and reloaded the config backup I had and it worked right away. This time I disconnected power and checked connections and it started working again after doing that twice. I'm not sure whats going on, the logs are available at: http://wikisend.com/download/440042/logs.zip

Help would be much appreciated.

Eric

ccrisan, 2015-06-23:
Hi,

It seems that the system no longer detects your CSI camera. How much RAM have you allocated to the camera? Also, check the ribbon cable and make sure it's connected properly.

EHIRL, 2015-06-23:
Hi again, 

Hmm that's interesting, I have not allocated any RAM to the camera, should I and if so how much? I changed the ribbon cable just encase but it looked perfect to me.

Eric

ccrisan, 2015-06-23:
Normally the default value is 128 MB for the camera. If you didn't touch anything in Expert Settings, the value should be 128 and should be enough. How about overclocking? Did you overclock your PI? Honestly I have no other idea about why your camera wouldn't work.

EHIRL, 2015-06-28:
It's working fine for the last couple of days. It is modestly over-clocked. I think it may have been the cable to the camera or the power supply, both were fine on Pi Projects before. Thanks for the help anyway!

closed 2015-06-28 by ccrisan