ccrisan/motionpie #230: Movies / Images Playback issue when multiple Cameras

SmartBlug, 2015-06-26:
If you've got several camera and want to have a look on files that have been recorded by one of the cameras (for ex Camera2), it shows all the files (recorded by all cameras) even if the title is correct ("Movies recorded by Camera2")

By the way, thanks for this great software !

ccrisan, 2015-06-26:
Have you configured the same media files directory for the both cameras?

SmartBlug, 2015-06-26:
Exact, my mistake.

That was not so obvious so may I suggest that in the default folder, you create a sub-folder with a guid or anything corresponding to the device (not the name that may change but anything computed on device creation) and then display only theses files for the device when selected.
May be you can add a small button for all files if we want to see all files from all devices.

Regards,

ccrisan, 2015-06-26:
Alright, I'll keep that in mind.

closed 2015-06-26 by ccrisan