ccrisan/motionpie #231: Lifetime max 4 hours with 2 cams

rv32, 2015-06-27:
Hi,
If use only the CSI PiCam only everything is perfect! When I add a low cost USB camera, it works fine during maximum 4 hours (between 1 & 4 in fact) & then the raspberry crashes.
To try to fix that I've set the GPU memory to 448 (max) without effect, raspi freeze after a few hours.

RV

ccrisan, 2015-06-28:
The usb camera (driver) may be buggy. Try with another usb camera or try the usb camera for 4+ hours with raspbian. I'm afraid there's not much I can do to help you with this problem.

ccrisan, 2015-06-28:
It would also help us if you mentioned the model of your camera (in the issue's title, preferably).

rv32, 2015-06-28:
Thanks for your reply. The problem is I don't know the model of this USB cam ! It's from "shenzen direct" at €2.5 !Absolutely no name on it. I was happy to make it work at the first time but not for a long time!

ccrisan, 2015-06-29:
Well in that case I'm sorry but I have no solution for your issue.

closed 2015-06-29 by ccrisan

rv32, 2015-06-29:
Is there a method to get the camera type even if there's nothing written on it?

ccrisan, 2015-06-29:
lsusb on a Linux machine (other than motionPie).