ccrisan/motionpie #232: Cam only function when a given MAC is not present on the LAN?

snagglewest, 2015-06-27:
So, I have several RP cams on the exterior of the house and they work perfect for monitoring the driveway and shed in the back.  I'd like to add a internal cameras inside the house now to check on the kids and pets when we aren't home.  The wife likes the idea but doesn't want them recording when she's home and to be honest I really don't either.  So I got to thinking would it be possible to automate the process and shut motion off if say a particular MAC address is attached to the LAN.  I found the link below and it seems they solved the problem with a "cron job".  But I know absolutely nothing about a cron job much less how to install it or whatever you do to make them work.  I was thinking it would be nice is MotionPie had the capability organically in the GUI.

In short - I'm wondering if you might consider adding the capability as an enhancement for MotionPie some day?  Maybe with the ability to add several different MACs to a list?

http://superuser.com/questions/493003/how-to-run-webcam-software-only-when-i-am-not-home-phone-is-not-on-the-lan

ccrisan, 2015-06-28:
It's a good idea that I might consider implementing in the future.

Squal78, 2015-07-05:
Hi all,
I think about an enhancement for MotionPie llike that.
For my side, I think it could be interresting to have API to active/desactivate detections. after, you can interface it with a cron or others...
Many thanks for this amazing software !

mastix, 2015-10-07:
I've created a simple script that I can run via SSH to start of stop the motion detection... :)

I saved it as "motion_detection_switch" and call it like "motion_detection_switch ON" or "motion_detection_switch OFF".

    #!/bin/bash
    
    if [ "$1" == "ON" ]
        then
            echo "Switching on motion detection..."
            sed -i -e "s/motion_detection off/motion_detection on/" /data/etc/thread-1.conf
        else
            echo "Switching off motion detection..."
            sed -i -e "s/motion_detection on/motion_detection off/" /data/etc/thread-1.conf
    fi

    /etc/init.d/S95motioneye restart