ccrisan/motionpie #233: Cannot update

rkenedi, 2015-06-28:
Hi there, have version 20141207 and when I click update I see there is a new version 20150601, but get this:

```
2015-06-27 20:10:23: ERROR: could not download update: Command '['/usr/bin/wget', u'https://github.com/ccrisan/motionPie/releases/download/20
2015-06-27 20:10:23: ERROR: Command '['/usr/bin/wget', u'https://github.com/ccrisan/motionPie/releases/download/20150601/motionPie-20150601.i
Traceback (most recent call last):                                                                                                          
  File "media/data/pi/motionPie/output/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1141, in _when_complete                 
  File "media/data/pi/motionPie/output/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1162, in _execute_method                
  File "/programs/motioneye/src/handlers.py", line 126, in wrapper                  
    return func(self, *args, **kwargs)                                                                                                      
  File "/programs/motioneye/src/handlers.py", line 1163, in post                    
    result = update.perform_update(version)                                                                                                 
  File "/programs/motioneye/src/update.py", line 168, in perform_update                               
    download(version)                                                                                                                       
  File "/programs/motioneye/src/update.py", line 104, in download                                                                           
    subprocess.check_call(['/usr/bin/wget', url, '--no-check-certificate', '-O', _DOWNLOAD_FILE_NAME])                                      
  File "/usr/lib/python2.7/subprocess.py", line 540, in check_call                                                                          
CalledProcessError: Command '['/usr/bin/wget', u'https://github.com/ccrisan/motionPie/releases/download/20150601/motionPie-20150601.img.gz',
2015-06-27 20:10:23: ERROR: 500 POST /update/?version=20150601 (10.0.1.17) 1018.67ms
2015-06-27 20:15:45: WARNING: Write error on 15: [Errno 110] Connection timed out
```

And I see that you have a newer version so perhaps the older one doesn't exist anymore? Am I stuck? Thanks!

ccrisan, 2015-06-28:
The url is aright and the file is still there. Try again later as the error indicates a network timeout problem.

rkenedi, 2015-06-28:
Thanks for the quick response. Are you sure? It's an error 500 and if I type this URL into my browser (which is connected):

https://github.com/ccrisan/motionPie/releases/download/20150601/motionPie-20150601.img.gz

I get a 404.

Thanks!

ccrisan, 2015-06-29:
Now I'm sure the link is *not* good :). The error should have been 404 in the first place. It's odd that it said "connection timed out". Nevertheless it seems that you are updating from a very old version, one that was still using the old path. Your only option is to manually save your data partition, download and write the new image and restore your data partition, again manually.

closed 2015-06-29 by ccrisan

rkenedi, 2015-06-29:
Ok cool thanks!