ccrisan/motionpie #234: Could there be support for Network File System (NFS) for linux only

dkelley810, 2015-06-28:
Calin,
I am wondering if NFS could be implemented to motionpie as another method of sharing files? I am working on converting from Samba to NFS as my sharing method

Couple of links i found to help: 
https://en.wikipedia.org/wiki/Network_File_System

http://tech.theswamp.in/post/2013/02/16/setting-up-a-raspberry-pi-powered-headless-media-server-and-nas/#comment-989072633

ccrisan, 2015-06-28:
Too few users would benefit from this feature that requires a considerable amount of work. Other features are more important these days.

dkelley810, 2015-06-28:
Ok Thank You! do you think I could implement it on my own? could I change install the packages needed on my own?

dkelley810, 2015-06-29:
Thank You! I will just use SAMBA!

closed 2015-06-29 by dkelley810