ccrisan/motionpie #236: Booting up with no Ethernet

khouadri, 2015-06-29:
Hi,
Astonishing achievement! I'm really admiring your work.

Issue:
I just installed Motionpie and booted it up with Wifi dongle only, Is it possible to work?
I've seen mp-... login: 
And the raspberry wasn't connected to the internet. 
What should I do pleaase ? Is ethernet necessary? :)

ccrisan, 2015-06-29:
You need to either boot it with ethernet the first time (and configure the wifi in the UI) or to write the image from Linux/OSX with [writeimage.sh](https://github.com/ccrisan/motionpie/wiki/Installation#write-the-os-image) with the -n agrument.

closed 2015-06-29 by ccrisan

khouadri, 2015-06-29:
Thank you so much for the quick response!! You didn't let me give up, this is my first project and I've been working all night to hopefully see it working.

I've seen the script page, and I was wondering if there is a detailed or similar tutorial for macOS to uderstand the steps?

ccrisan, 2015-06-29:
The tutorial should work as well on OSX. Other OSX users have managed to get it working with just installing some ext4 drivers. I wouldn't know exactly as I don't own an OSX machine.

khouadri, 2015-06-29:
Thank you so much