ccrisan/motionpie #237: Can't show image from usb camera

servantos, 2015-07-06:
/dev/v4l/by-id/usb-093a_2620-video-index0
lsusb show: Bus 001 Device 005: ID 093a:2620

The usb cam is fully work. Tested in another raspberry pi motion detect project.

ccrisan, 2015-07-06:
You provided way to little information. See https://github.com/ccrisan/motionpie/wiki/Reporting-An-Issue.

servantos, 2015-07-06:
I'm sorry.
All logs : http://s000.tinyupload.com/download.php?file_id=02328878403261632731&t=0232887840326163273153378

ccrisan, 2015-07-06:
Hi,

The only thing that looks suspicious is that "gspca_pac7302 1-1.4:1.0: URB error -71, resubmitting" message that keeps appearing in dmesg.log. There seems to be a lower level problem (USB or driver-related).

1. Is your power supply powerful enough?
2. Have you changed any expert settings (such as overclocking)?
3. Have you tried raspbian + [motionEye](https://bitbucket.org/ccrisan/motioneye/)?

closed 2015-08-09 by ccrisan