ccrisan/motionpie #238: Memory problem on raspberry pi ??

vrolland78, 2015-07-07:
Hi,
I encounter a problem with my motionpie version 20150601 running on a raspberry pi with a raspberry pi camera module.
After 2 or 3 days, the system is still running, the web interface is up but the video is down.

Here the dmesg...if you have any idea about the problem...I take it ?
Thanks,
regards
Vincent

[165203.045791] S35wifi invoked oom-killer: gfp_mask=0x201da, order=0, oom_score_adj=0
[165203.051223] CPU: 0 PID: 7010 Comm: S35wifi Not tainted 3.18.12-quick #1
[165203.058079] Backtrace:
[165203.072777] Function entered at [<c0011474>] from [<c001167c>]
[165203.078833]  r6:00000000 r5:c09ac000 r4:c09ac000 r3:00000000
[165203.085876] Function entered at [<c0011664>] from [<c04107e8>]
[165203.092401] Function entered at [<c04107c8>] from [<c040eb8c>]
[165203.094341] Function entered at [<c040eb10>] from [<c006d7c4>]
[165203.101389]  r10:c04ee418 r9:c04ee550 r8:0000f879 r7:00000000 r6:000201da r5:000201da
[165203.106217]  r4:cf9ee400
[165203.112672] Function entered at [<c006d534>] from [<c006df1c>]
[165203.114681]  r10:c04ee418 r9:c04ee550 r8:0000f879 r7:00000000 r6:000201da r5:00000000
[165203.120301]  r4:00000000
[165203.122524] Function entered at [<c006dc24>] from [<c0071a74>]
[165203.125940]  r10:00000001 r9:00000000 r8:c05160c0 r7:00000000 r6:c04ec0f0 r5:000201da
[165203.130191]  r4:00000000
[165203.133193] Function entered at [<c007121c>] from [<c006c428>]
[165203.135296]  r10:cc9340e0 r9:00000084 r8:c09addc8 r7:cf4a262c r6:cf4a262c r5:cf9ca540
[165203.141235]  r4:00000084
[165203.143414] Function entered at [<c006c18c>] from [<c008ac34>]
[165203.145467]  r10:ca340000 r9:00000084 r8:c09a8000 r7:c09ac000 r6:0008c000 r5:cc9340e0
[165203.155860]  r4:ca340200
[165203.159079] Function entered at [<c008abf8>] from [<c008d24c>]
[165203.169985]  r4:ca340200
[165203.174570] Function entered at [<c008d068>] from [<c008ded0>]
[165203.176662]  r10:000000a8 r9:0000008c r8:c09ac020 r7:cc9340e0 r6:c09a8000 r5:0008c000
[165203.184249]  r4:00000000
[165203.186529] Function entered at [<c008d6e0>] from [<c00140e4>]
[165203.191858]  r10:c09adfb0 r9:cfa567f4 r8:0008c870 r7:cfa65860 r6:80000007 r5:cfa567c0
[165203.196954]  r4:c09ac038
[165203.199063] Function entered at [<c0013f98>] from [<c00084c0>]
[165203.201108]  r10:00000000 r9:000bab30 r8:c09adfb0 r7:0008c870 r6:c04ef744 r5:c0013f98
[165203.207204]  r4:00000007
[165203.209340] Function entered at [<c0008484>] from [<c0012680>]
[165203.212151] Exception stack(0xc09adfb0 to 0xc09adff8)
[165203.214231] dfa0:                                     00000000 00000054 00000000 013f3388
[165203.219245] dfc0: 013f32e8 013f33c8 013f33c8 00000000 00000000 000bab30 00000000 00000000
[165203.223639] dfe0: 013f33b2 beaa0900 000541c0 0008c870 80000010 ffffffff
[165203.225787]  r8:00c5387d r7:00c5387d r6:ffffffff r5:80000010 r4:0008c870
[165203.228124] Mem-info:
[165203.231647] Normal per-cpu:
[165203.233873] CPU    0: hi:   90, btch:  15 usd:  59
[165203.235919] active_anon:54904 inactive_anon:34 isolated_anon:0
                 active_file:140 inactive_file:164 isolated_file:0
                 unevictable:3226 dirty:0 writeback:2 unstable:0
                 free:604 slab_reclaimable:216 slab_unreclaimable:589
                 mapped:3024 shmem:40 pagetables:239 bounce:0
                 free_cma:0
[165203.249233] Normal free:2416kB min:2016kB low:2520kB high:3024kB active_anon:219616kB inactive_anon:136kB active_file:596kB inactive_file:708kB unevictable:12904kB isolated(anon):0kB isolated(file):0kB present:262144kB managed:254436kB mlocked:12904kB dirty:0kB writeback:8kB mapped:12124kB shmem:160kB slab_reclaimable:864kB slab_unreclaimable:2356kB kernel_stack:472kB pagetables:956kB unstable:0kB bounce:0kB free_cma:0kB writeback_tmp:0kB pages_scanned:7920 all_unreclaimable? yes
[165203.262151] lowmem_reserve[]: 0 0
[165203.264473] Normal: 88*4kB (UEMR) 8*8kB (MR) 1*16kB (R) 0*32kB 1*64kB (R) 1*128kB (R) 1*256kB (R) 1*512kB (R) 1*1024kB (R) 0*2048kB 0*4096kB = 2416kB
[165203.270411] 1348 total pagecache pages
[165203.290526] 65536 pages of RAM
[165203.293969] 770 free pages
[165203.296176] 1927 reserved pages
[165203.298573] 677 slab pages
[165203.307635] 270884 pages shared
[165203.309674] 0 pages swap cached
[165203.314780] [ pid ]   uid  tgid total_vm      rss nr_ptes swapents oom_score_adj name
[165203.329151] [   62]     0    62      501      241       4        0             0 tee
[165203.340787] [   79]     0    79      501      307       5        0             0 syslogd
[165203.349791] [   82]     0    82      501      240       5        0             0 watchdog
[165203.361950] [   86]     0    86      731      324       5        0         -1000 udevd
[165203.368839] [  414]     0   414     1089      446       5        0             0 wpa_supplicant
[165203.378815] [  425]     0   425      738      212       5        0             0 S35wifi
[165203.388813] [  442]     0   442      578      331       6        0             0 dhclient
[165203.406072] [  443]     0   443      743      217       5        0             0 S40network
[165203.417837] [  474]     0   474     1130      515       5        0         -1000 sshd
[165203.424048] [  563]     0   563     3552     3227      11        0             0 ntpd
[165203.432864] [  570]     0   570      501      218       4        0             0 crond
[165203.438841] [  589]     0   589    50026    48534     101        0             0 python2.7
[165203.442823] [  652]     0   652      738      231       3        0             0 S95motioneye
[165203.446831] [  655]     0   655      502      308       4        0             0 getty
[165203.451637] [ 6074]     0  6074    21636     8665      37        0             0 motion
[165203.454990] [ 7005]     0  7005     1130      707       5        0             0 sshd
[165203.458206] [ 7006]     0  7006     1048      353       5        0             0 wget
[165203.466824] [ 7007]     0  7007      158        2       3        0             0 ip
[165203.475816] [ 7008]     0  7008      158        2       3        0             0 ip
[165203.480324] [ 7009]     0  7009      743      238       5        0             0 S40network
[165203.484241] [ 7010]     0  7010      738      183       5        0             0 S35wifi
[165203.488497] Out of memory: Kill process 589 (python2.7) score 741 or sacrifice child
[165203.492623] Killed process 6074 (motion) total-vm:86544kB, anon-rss:25308kB, file-rss:9352kB
[165204.513832] bcm2835-v4l2: error 0 waiting for frame completion

ccrisan, 2015-07-07:
How much ram does your device have? What resolution do you use? How many cameras do you use? Only the PI camera module?

vrolland78, 2015-07-09:
Hi ccrisan,
Thank you to take the time to look at my problem.
My settings : 
In "Expert Settings"
	GPU Memory : 256
	Overclocking : modest
In "Video DEvice"
	Video Resolution : 1280x1024
	Frale Rate : 20

I only have 1 camera on this raspberry (Camera Device : /dev/video0 - Camera Type : V4L2 Camera)
Video Streaming is Off, motion Detection and Notifications are On

Actually, I was looking at the settings for writing my answer and python crash again, last reboot 2 days ago

[160866.068302] Out of memory: Kill process 588 (python2.7) score 737 or sacrifice child
[160866.071794] Killed process 625 (motion) total-vm:88468kB, anon-rss:25540kB, file-rss:9452kB
[160867.092056] bcm2835-v4l2: error 0 waiting for frame completion

I dit not have any problem with the previous version (20150516)

ccrisan, 2015-07-09:
Well clearly your device ran out of memory. I couldn't say why this did not happen with the previous version. However, I suggest either setting 128 MB for the GPU (your camera normally doesn't need more than that) or decreasing the resolution. Also, make sure that there aren't too many media files that piled up on your SD card, as they are regularly scanned.

closed 2015-08-09 by ccrisan