ccrisan/motionpie #239: Video Stream comes and goes.

rockman55, 2015-07-08:
I have two Pi 2 B+ currently running on my system when I log in either as admin or user I can see the video feed just fine from both cameras, but in just a few seconds the feed goes away and is replaced by the No Camera Icon. Then the feed returns, this is mostly annoying but I sure there must be some setting that I have incorrectly set. If you could tell me what logs to post I will do promptly just not sure what you might need to see.
Thanks,
Danny

ccrisan, 2015-07-08:
Hi,

Please send me *all* the logs. See [Reporting An Issue](https://github.com/ccrisan/motionpie/wiki/Reporting-An-Issue) for more details.

rockman55, 2015-07-08:
I am sending the logs as a reply to this email. Tried to post them on Wiki and they needed to be png or jpg.

Thanks,

Danny

ccrisan, 2015-07-09:
Well? Did you send the email? I haven't received them or I don't know where to look for them. Also, the wiki page I pointed you to does not say that you should attach the logs to the issue directly, but suggests that you use a small files sharing service.

rockman55, 2015-07-09:
The logs are attached to this email.

Danny

ccrisan, 2015-07-09:
I'm sorry but there are no logs attached to the email. Please either send me a regular email (not via github, as it seems to remove any attachments) or follow the instructions from the wiki page I gave you.

rockman55, 2015-07-09:
Let’s try this again, I downloaded these log files using tinyupload.com. Let me know if this works.

I had been attaching the files to the email messages that I got from you, I guess that like you said github stripped them off.

Thanks,

Danny

 

This is link for boot.log:  http://s000.tinyupload.com/index.php?file_id=53513204048612763400

This is link for mseg.log:  <http://s000.tinyupload.com/index.php?file_id=02422476553099820173> http://s000.tinyupload.com/index.php?file_id=02422476553099820173

This is link for messages.log:  <http://s000.tinyupload.com/index.php?file_id=06885190588306174238> http://s000.tinyupload.com/index.php?file_id=06885190588306174238

This is link for motion.log:  <http://s000.tinyupload.com/index.php?file_id=53513204048612763400> http://s000.tinyupload.com/index.php?file_id=53513204048612763400

This is link for motioneye.log:  <http://s000.tinyupload.com/index.php?file_id=32079594286282223123> http://s000.tinyupload.com/index.php?file_id=32079594286282223123

ccrisan, 2015-07-10:
Hi,

I received the files and I'm seeing something weird in your motion.log file: `conf_cmdparse: Unknown config option "1280x800"`. I would also like to see your `/data/etc/motion.conf` and `/data/etc/thread-1.conf` files. If you don't know how to get those, just hit the *Backup* button next to *Configuration* and it'll download the entire /data/etc directory on your PC. Make sure you remove any sensitive data (i.e. passwords) from the files you send me.

Calin.

rockman55, 2015-07-10:
Calin I am using tinyupload.com again for the two files you requested. The thread-1.conf is in its original text state. On the motion.conf file it is a screen shot that I redacted the passwords. Thanks again,

Danny

 

Link for motion.conf file : http://s000.tinyupload.com/index.php?file_id=00066021137919536977

Link for thread-1.conf file : http://s000.tinyupload.com/index.php?file_id=50612178291329051350

ccrisan, 2015-07-12:
You forgot to attach the thread-3.conf file. That's the one that seems to cause the issue.

rockman55, 2015-07-13:
Thanks Calin I did not know that you requested this file. Here it is:

Link for thread-3.conf: http://s000.tinyupload.com/index.php?file_id=32053800313507842226

 

Danny

ccrisan, 2015-07-13:
You seem to have a 3rd camera added and enabled (the 2nd one was disabled). The last line in thread-3.conf is 1280x800 which... to he honest I have no idea how it got there. Remove the thread-3.conf line from motion.conf or simply remove all cameras and re-add them all and let me know if the problem persists.

rockman55, 2015-07-13:
Thanks Calin, I think this has resolved my issue. I deleted all the cameras and then added them back. I am not getting the no camera logos. One additional question please. If I want to set my cameras on the Fast Network Camera option, can I have all three cameras set to this option? 

Thanks,

Danny

ccrisan, 2015-07-14:
I'm glad I could help. Nope, you can only have one Fast Network Camera and that must be the Rasperry PI camera board.

closed 2015-07-14 by ccrisan

rockman55, 2015-07-14:
Thanks Calin you have been a great help.

https://github.com/ccrisan/motionpie/issues/239#issuecomment-121144588