ccrisan/motionpie #240: An error occurred. Refreshing is recommended

rpilover, 2015-07-13:
I had the Motionpie running all last week and each day I checked the images and they were fine. I left it running over the weekend and now when I try accessing the images i get "An error occurred. Refreshing is recommended" I have tried rebooting lots of times and I have tried accessing from different browsers and still can't access the images.

ccrisan, 2015-07-13:
https://github.com/ccrisan/motionpie/wiki/Reporting-An-Issue

rafi1975, 2015-07-15:
Hi,

I have similar issue - error - "An error occurred. Refreshing is recommended". It happens everytime I want to add another network camera - (one RPI NoIR and another USB are already connected to RPI 2). It was ok when I added it for the first time. I can see the following errors:

 015-07-15 10:29:14: INFO: adding new camera with id 3...
2015-07-15 10:29:14: ERROR: cannot concatenate 'str' and 'NoneType' objects
Traceback (most recent call last):
  File "home/ccrisan/motionpie/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1288, in _stack_context_handle_exception
  File "home/ccrisan/motionpie/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1475, in wrapper
  File "/programs/motioneye/src/handlers.py", line 197, in post
    self.add_camera()
  File "/programs/motioneye/src/handlers.py", line 121, in wrapper
    return func(self, *args, **kwargs)
  File "/programs/motioneye/src/handlers.py", line 618, in add_camera
    camera_config = config.add_camera(device_details)
  File "/programs/motioneye/src/config.py", line 507, in add_camera
    set_camera(camera_id, camera_config)
  File "/programs/motioneye/src/config.py", line 428, in set_camera
    lines = _dict_to_conf(lines, camera_config)
  File "/programs/motioneye/src/config.py", line 1450, in _dict_to_conf
    line = name + ' ' + _python_to_value(value)
TypeError: cannot concatenate 'str' and 'NoneType' objects
2015-07-15 10:29:14: ERROR: 500 POST /config/add/?_=1436952556338&_username=admin&_signature=9332e4558df58f0ae3164fed5941be278c878f06 (178.167.254.167) 26.96ms
2015-07-15 10:29:15: ERROR: failed to create movie preview for /data/output/2015-07-13/00-34-52.avi: Command 'ffmpeg -i "/data/output/2015-07-13/00-34-52.avi" -f mjpeg -vframes 1 -ss 4 -y /data/output/2015-07-13/00-34-52.avi.thumb' returned non-zero exit status 1

ccrisan, 2015-07-15:
@rafi1975 what version of motionPie? Is your problem fully reproducible?

rafi1975, 2015-07-15:
The latest - 20150601
It's been happening everytime I try to add any network motion pie camera.

ccrisan, 2015-07-15:
@rafi1975 could you please update to the prerelease 20150621 version and see if the problem persists? You need to activate the *Enable Prereleases* option in *Expert Settings*. After upgrading, please clear your browser's cache as some javascript files have also been changed.

rafi1975, 2015-07-16:
I just did that. However what I've noticed is that when providing the admin's password and hitting 'Enter' gives that error.
I noticed by accident that if I enter the password and change the 'password' field using Tab, it gives me the list of camera/s attached to the remote motionpie. Then it all works ok.
However, another issue has been spotted. With one RPI camera and another USB attached to the same RPI22 after a reboot only the USB camera is present. Performing several reboots doesn't cure the situation at all.
I had to disconnect USB camera, then reboot RPI and once the onboard camera was detected I had to connect the USB camera to PI and it was detected ok.
It can be reproduced.

Thanks,

Rafi

ccrisan, 2015-07-16:
1) So did the problem as originally described go away after upgrading or not?
2) Hitting enter in the Add Camera dialog may lead to unpredictable problems indeed. I'll need to investigate that a bit.
3) What do you mean by "only the USB camera is present"? Does it disappear from the UI? Did you investigate it at lower levels and saw that the video device file is actually missing from the system? Could you provide me with the log files right after a boot with camera missing?

P.S. next time you encounter a problem that is different from the one discussed so far in one issue here, please create a new issue, as it's painfully difficult for me to keep track of all the issues and their names if multiple problems are described in one github issue. Thanks :)

rafi1975, 2015-07-17:
Hi,

Replying to your questions:

1) So did the problem as originally described go away after upgrading or
not?

Hard to say it it went away after the upgrade. As I already mentioned I
noticed that when changing the password field by using 'Tab" key I was
presented with a drop-down list with available remote cameras. I have no
other Pi available to check with the previous OS release.

2) Hitting enter in the Add Camera dialog may lead to unpredictable
problems indeed. I'll need to investigate that a bit.

Yup. I noticed that by accident - luckily and got the cameras added ok.

3) What do you mean by "only the USB camera is present"? Does it disappear
from the UI? Did you investigate it at lower levels and saw that the video
device file is actually missing from the system? Could you provide me with
the log files right after a boot with camera missing?

I haven't tried with lower levels TBH. But as far as I can say it happens
all the time the PI is rebooted. The on-board PI camera doesn't show
anything - only the USB one is ok. Once the USB is disconnected from USB
hub or Pi itself, the onboard starts working ok. Then I have to plug the
USB again and all is flying.

Below are links to all log files:

http://s000.tinyupload.com/index.php?file_id=65988766427030110390
http://s000.tinyupload.com/index.php?file_id=11061143733993517389
http://s000.tinyupload.com/index.php?file_id=07178717857652047263
http://s000.tinyupload.com/index.php?file_id=81701729856930735009
http://s000.tinyupload.com/index.php?file_id=44197877118052091589

Thanks,

Rafi

ccrisan, 2015-07-17:
@rafi1975 I suspect a power problem there, especially since you're talking about a USB hub. It's possible that recent versions eat up a bit more power than the previous ones (more features, different overclocking settings) - it's hard to tell. In any case, I believe it's a hardware problem rather than a software one. Please experiment a bit by changing various parts of the hardware.

rafi1975, 2015-07-17:
Ok, will do. The hub is powered externally, not from PI.