ccrisan/motionpie #241: MotionPie - saving videos

awoodman, 2015-07-14:
Hi guys,

I'm fairly new to Raspberry PI and new to MotionPie.  I successfully installed MotionPie on my RaspPi B+, and was able to access the GUI on a separate PC and get video through the local network.  I'm wondering what the best was to store video, on a remote server or FTP.  I also want to be able to access the GUI and camera outside of my local network.  I realize I need to use port forwarding on my router, and I'm still working on that.  What is the best way to store the video footage, as I've mentioed?  Thank you!  

Also, I was successful at plugging ethernet cable into Raspberry PI, and then turning on WiFi, but the wifi eventually times out and does not reconnect.  I searched around and saw this is an ongoing issue, so for now my plan is to just use the ethernet.

ccrisan, 2015-07-14:
1) Videos can be saved either locally, on your SD card, on a USB drive or on a network share via SMB. FTP support is not yet available. The best option depends on your use case.
2) Port forwarding is indeed a task that only you can do.
3) WiFi issues are possible but there's not much I can do about them. Make sure you use a good power supply, a good adapter and a good wireless router. Also make sure your PI is not too far from the router.

awoodman, 2015-07-14:
Thank you for the help.  So if I understand correctly, I cannot save the video onto anything but hardware that is accessible in my LAN?  Here's my scenario:  I set up the motionpie and leave town for a week, and someone breaks into my house and takes the PI.  Is there a way to store the video somewhere outside of my LAN, so that I still have the footage even if the hardware got stolen?  Is this what you're referring to, by SMB?

ccrisan, 2015-07-15:
That is correct, there's currently no easy way to save the media files somewhere out of your LAN. It's been planned for a while and it will hopefully land soon in motionEye/motionPie, but don't hold your breath.

BettySwallocks, 2015-07-17:
I adapted someone else's short bash script (sorry I can't find the original author) to trigger an FTP upload of my relevant files after a motion detection. This can be used under the MotionPie 'Motion Notifications', 'Run A Command' area:-

    #!/bin/bash
    #
    # wait in case motion trigger fires before pic/vid files written locally
    sleep 30
    #
    # explicit directory set (needed in my case when run from MotionPie)
    cd /data/output
    #
    # get today's date using %F (same as %Y-%m-%d) - check your date standard!
    now=$(date +'%F')
    #
    # output date value for logging (motion.log)
    #echo $now
    #
    # output present working directory for logging (motion.log)
    #pwd
    #
    # find just files with today's date in them
    # combined with exec of curl command, per object
    # curl set to upload to my ftp target
    find $now -type f -exec curl --user <youruser>:<yourpassword> --url ftp://<yourFTPserver>/<youroptionalsubdir>/ --upload-file {} \;
    #
    # remove all source files just transferred with recursive and force options
    rm -rf $now

ccrisan, 2015-07-17:
@BettySwallocks thanks for your contribution!

BettySwallocks, 2015-07-17:
Minor tweaks to deal better with logging and to ensure files go into daily sub-directories on target FTP server.

    #!/bin/bash
    #
    # wait in case motion trigger fires before pic/vid files written locally
    sleep 30
    #
    # explicit directory set (needed in my case when run from MotionPie)
    cd /data/output
    #
    # get today's date using %F (same as %Y-%m-%d) - check your date standard!
    now=$(date +'%F')
    #
    # output date value for logging (motion.log)
    #echo $now
    #
    # output present working directory for logging (motion.log)
    #pwd
    #
    # create a remote directory for daily files (if needed)
    # (if already there then 550 error, which can be ignored)
    curl -Q "MKD ipcam/$now" --user <youruser>:<yourpassword> --url ftp://<yourFTPserver> --silent
    #
    # find just files with today's date in them
    # combined with exec of curl command, per object
    # curl set to upload to my ftp target
    # write-out and silent keep output to just useful file level summaries
    find $now -type f -exec curl --write-out "Sent To: %{remote_ip}  Response code: %{response_code}  Total time: %{time_total}  Size: %{size_upload}\n" --silent --user <youruser>:<yourpassword> --url ftp://<yourFTPserver>/ipcam/$now/ --upload-file {} \;
    #
    # remove all source files just transferred with recursive and force options
    rm -rf $now

mixmediasec, 2015-07-26:
Is there a way to view live stream outside of the network the my pi is on?

scohrab, 2015-08-28:
One way, which works for me, is to use an old laptop to provide an SMB share to MotionPie. Configure the laptop with Microsoft's OneDrive (or Dropbox, or whatever other filesync product you fancy), and make sure the SMB share is sync'd by your chosen product. You can then see your files both locally and remotely, without needing to get into port forwarding.