ccrisan/motionpie #243: Stopping motion via script

victorbrca, 2015-07-18:
Hello,

Is there a way to disable/enable the motion binaries via script? I'm trying to setup a script on my router that SSHs into motionPie and disables the motion monitoring when my cellphone is connected to the wifi.  

Thanks,
Victor.

ccrisan, 2015-07-18:
Well, not yet. But given the large number of users that asked for such a feature, it's on my short todo list.

victorbrca, 2015-07-19:
Is this something that can be done with the original version of motioneye (if I manually install it in Raspbian)? 

Another option I'm thinking would be to stop motioneye completely using the startup script (S95motioneye).

ccrisan, 2015-07-19:
Nope, there's nothing more to motionEye that cannot be done with motionPie; it's usually the other way around. The motion detection control can be done via an HTTP request to the motion daemon. I'll need to make that available via a "public motionEye API" some day.

ccrisan, 2015-07-19:
Duplicate of https://github.com/ccrisan/motionpie/issues/232.

closed 2015-07-19 by ccrisan

mastix, 2015-10-07:
I've created a simple script that I can run via SSH to start of stop the motion detection... :)

I saved it as "motion_detection_switch" and call it like "motion_detection_switch ON" or "motion_detection_switch OFF".

    #!/bin/bash
    
    if [ "$1" == "ON" ]
        then
            echo "Switching on motion detection..."
            sed -i -e "s/motion_detection off/motion_detection on/" /data/etc/thread-1.conf
        else
            echo "Switching off motion detection..."
            sed -i -e "s/motion_detection on/motion_detection off/" /data/etc/thread-1.conf
    fi