ccrisan/motionpie #246: Issue with initial boot.

JesterLestat, 2015-07-21:
So I downloaded the image, unpacked it, imaged it to the MicroSD, plugged it in, everything seemed to work, other than one problem. I can't go through the initial setup of a camera, updating anything, upgrading anything, because I can't log on. There seems to be a lack of default username and/or password in any of the "installation" instructional articles anywhere online. I can log in to the Pi from a browser, but that doesn't do me any good if I can't expand the file system, set up the Pi for wireless (unless I want to try to monitor my driveway through 2 walls and a good 10-20 yards), enable the camera, enable SSH, or anything else required for setup. I saw somewhere that it initially sets the root password to the board number, ironically apparently where to find you board number, without being able to log in to the Pi and checking through commands, seems to be an extremely poorly documented topic. It might help if you used a standard username/password set up like root/password or pi/raspberry and just have an instructional for how people can change the default password. I've been using Linux since high school (14+ years ago), fairly new to the Pi, but never have I had such a problem finding initial log in information for a fresh install of any Linux distro. You need to work on the initial installation instructions.

ccrisan, 2015-07-21:
Had you have been using Linux for such a long time, your RTFM skills would have been a lot better. You would have also known that this kind of attitude is not appreciated in FOSS communities. So I take it you're actually a Linux newbie; welcome to the Linux community!

You can start by heading over to motionPie's [wiki](https://github.com/ccrisan/motionpie/wiki) and taking a look over the 15+ wiki pages.

The [Installation](https://github.com/ccrisan/motionpie/wiki/Installation) page will walk you through the... well, the installation process, using one of the 3 common operating systems. The [First Boot](https://github.com/ccrisan/motionpie/wiki/Installation#first-boot) article will tell you that, after finding your motionPie's IP address, you need to use `admin` with no password when logging in on the web UI.

The [FAQ](https://github.com/ccrisan/motionpie/wiki/FAQ) page will help you dealing with [your board's serial number](https://github.com/ccrisan/motionpie/wiki/FAQ#whats-my-boards-serial-number) and with [Shell Access](https://github.com/ccrisan/motionpie/wiki/FAQ#shell-access).

The [Tweaks](https://github.com/ccrisan/motionpie/wiki/Tweaks) page will allow you to dig even deeper into the system's internals.

motionPie will automatically:
 * expand your data file system
 * enable and configure any connected cameras
 * configure wired network using DHCP
 * configure and start SSH and FTP servers

The only thing you need to do is configure your wireless network, which is done from the UI, using a temporary wired connection.

P.S. you're welcomed.

closed 2015-07-21 by ccrisan

JesterLestat, 2015-07-21:
Actual my RTFM skills are just fine, but it is standard practice with open source operating systems that you document and list the primary intended default user's password for local access for people who have a need for quick access to perform more advanced and/or special settings, that may not be covered in your forums, with the installation steps, or at least provide a link to where said information can be easily obtained. As far as my attitude, I do apologize for my attitude, but everything that I said is still true about the poor manner of documentation. By the way, I love the condisending way you straight up call me a liar in response to me pointing out the flaws in your documentation for initial install. I guess in similar fashion I should also assume and make public note on this forum that this must be your first time creating, documenting and supporting your own linux based distro. Welcome to the world of open source support back at you newbie.

Now for sale of argument let's go ahead and run with your assumption that someone who doesn't confirm to politities to protect your feels when you miss a huge part of providing documentation for your software must be a newb and a liar, and therefore must not also work in IT where they provide and maintain the technical documentation for open source and proprietary software at an internationso company with more than 150,000 employees. Lets also assume that person isn't responsible for providing the documentation for a contingency plan on how to recover the system in a worse case scenario, such as the network is down, so the webpage doesn't load, and somehow a tech has to log in to the camera locally with a restored image. All that information, such as how to find the serial number would have been helpful if it were mentioned or at least linked to on the initial install tutorial page. In certain situations you can't just assume people have the time or the knowledge on how to navigate your entire support forum.

Also out of the bullet points you provided your initial boot page does not say anything about it setting the Wi-Fi for DHCP. Might want to add that documentation into the list of steps the image takes in those 1-2 minutes.

Again I apologize for my attitude, but at the very least the initial boot documentation should not just say the password is set to the board's serial number, but also state where to find that number or provide a link to the FAQ for that. If logging in locally after the initial boot is a frequently asked question that should tell you a lot about your documentation.

ccrisan, 2015-07-21:
I wasn't calling you a liar, I was just being ironic. Should I point you to the [word's reference](https://en.wiktionary.org/wiki/ironic) or can you find it yourself?

I wonder how 2800+ users (according to github) managed to get their motionPies running, probably without having as many years of Linux experience as you say you have. It's funny how people who are new to the FOSS world know how to appreciate a man's work (that has been done in his spare time), while a pretended long-time Linux user makes a big deal out of some documentation issues, without even saying *thanks*. Man, you're one in 2800 - you're almost unique!

The documentation is far from perfect, but is alright; you can usually find what you're looking for in a matter of minutes. It's a work in progress, just like the entire project is. But with not-so-constructive criticism such as yours I feel like losing the will to spend my time on it. What I would do in this case, after 14 years of  Linux experience, is to come up with a better version of the documentation (it's git-based, by the way) and politely suggest merging my changes into the official one, without bragging about my Linux experience.

I won't accept your apologies; the attitude is still there with your second comment, while the *thanks* is still missing. Not to mention a donation which a lot of the users have made for the project. But you, not working for a company with more than 150,000 employees, wouldn't be able to afford it (don't even think of making it now, as I will reject it).

JesterLestat, 2015-07-21:
Actually I was making that post to point out the flaws in your documentation for initial installation. Because a good deal of that information on logging in locally isn't very transparent for new users that are just trying to follow a quick installation guide to drop a camera in and run, or for people who may be using your product, in a situation similar to mine, such in case of a worst case scenario where that information may be vital for immediate restoration of operations. We had our cameras up and running within an hour. And no I haven't made a donation, because I'm not the one funding the testing of the open source security cameras. Should the demonstration go smoothly enough that the board's or their ad odors agree to continue, the company may move on to a testing environment phase, if that also runs smoothly, they'll go to a live facility testing, and then may move to a full implementation. I was just trying to get a head start on recording the documentation for the testing phases, and noticed the issues with your documentation. If the company decides to progress with your software past the demo, then it will be up to their discretion as to when to donate to your project or how much. Though I love the pressure to donate or shut up. 

Yes, you were totally being "ironic" by attempting to insult my intelligence and bring in to question my background and knowledge of Linux, which you continue to do in your second post.

You're right however that I didn't change my way of saying things, I didn't conform to who you want me to be. Yes, I know I am abrasive. I apologized if it offended you, not for being me. You will not get an apology for the way that I say things, but whether you accept the apology for me coming off abrasive and offensive to you is on you. I made the effort on my part, it's not my business if you can't forgive people. Like they say, "you can lead a horse to water, but you can't force it to take a drink." I'm just trying to help you with your documentation, to make it easier to follow. I think you have a great product, but you seem to have made the assumption that a new user isn't going to see that the password sets to the serial number and instantly pull their project apart look all over the raspberry pi for the number then try to start over.

JesterLestat, 2015-07-21:
Oh, and you may not have directly called me a liar, but you heavily I sinuated it with your passive aggressive "ironic" statement about me being a newb and the welcoming me to the world of Linux that followed.

ccrisan, 2015-07-21:
I never really doubted your intelligence; it's just that you don't seem to put it to good use. Instead of writing all these literally useless comments, you could have spent your time reorganizing/rewriting my "poor" documentation.

Unless you're willing to contribute with some actual work, please stop commenting here.

JesterLestat, 2015-07-21:
It was called advise. To some people having someone with experience in a matter such as technical documentation offer advise is considered a contribution. I could rewrite it for you, if and when I have time, but as my time is typically pretty limited I figured I'd just make the offer that you might want to include some information in your documentation or at least reorganize it to make it a bit more user friendly to newer users.