ccrisan/motionpie #247: False Positives with brightness change

SuperSharpShooter, 2015-07-21:
Every time the environmental light changes slightly (sun goes behind a cloud etc), the camera is falsely triggered. This happens upwards of 30 times a day, even after increasing the frame change threshold to a point where it no longer detects real motion. Is there a way to ignore these changes in brightness that the whole frame sees, and allow triggering only when real motion is detected?

ccrisan, 2015-07-22:
Make sure you have the *Light Switch Detection* enabled. What camera do you use?

SuperSharpShooter, 2015-07-22:
I do have light switch detection enabled. I'm using the rpi cam.

Here's a small example of a video it creates. https://www.dropbox.com/s/1489zxxuuek7z7h/11-36-22.avi?dl=0

ccrisan, 2015-07-22:
People have been complaining about this brightness change problem when working with the PI camera. I still need to investigate and to find the root cause of these sudden brightness changes. You don't have the automatic brightness option enabled, do you?

ccrisan, 2015-07-22:
Here's a thread that treats this topic: https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=62364&start=325.

For now, there are three possible workarounds for your problem:
1. Use the `/usr/bin/motion-mmal` binary instead of `/usr/bin/motion` in *Expert Settings*.
2. Use `bcm2835-v4l2 max_video_width=2592 max_video_height=1944` in `/etc/modules`, if you're using an older version of motionPie. New versions already have these options enabled.
3. Use the [Fast Network Camera][https://github.com/ccrisan/motionpie/wiki/Fast-Network-Camera] mode. You'll lose many of the features (such as motion detection!) but you get full control over the PI camera options.

Let me know if one of these workarounds works for you.

SuperSharpShooter, 2015-07-22:
I'm running 20150621 on a Rpi2, so the first two workarounds don't apply. I don't have automatic brightness enabled, as this just cycles continuously from light to dark.

The sole purpose of this setup is as a security camera, so losing motion detection isn't going to work in this case unfortunately.

ccrisan, 2015-07-23:
Why doesn't the first one apply? That was the one that I was hoping would help you.

SuperSharpShooter, 2015-07-23:
Sorry I thought running the new version of motion meant needing to use motion-mmal was redundant, but I will try it tomorrow and see if it works better. Thanks.

SuperSharpShooter, 2015-07-24:
Changing to the motion-mmal binary looks to have improved the situation. Haven't had any false positives from the first half of today, so hopefully it stays that way!

ccrisan, 2015-07-24:
I might make the motion-mmal binary the default when using a RPi camera, if everything works fine. I haven't gotten too much feedback about this motion-mmal binary though, so I'm still waiting for opinions. Your feedback is highly appreciated.

robmuller, 2015-07-24:
Motion-mmal seems to be working fine.
An advantage is low light performance. With standard motion exposure times are not influenced by the number of fps. With motion-mmal exposure times are longer when needed, up to 1/fps. Motion blur will happen ofcourse in low light (with "-ex night" and low fps) but that is much better than no motion detected at all.

SuperSharpShooter, 2015-07-27:
Today the light is more variable here, and I'm still getting quite a few false positives even with the motion-mmal binary with frame change threshold up to 6%.

SuperSharpShooter, 2015-08-06:
After more testing, for the best results (least false positives) in variable lighting conditions, auto noise detection must be disabled, and the manual noise level setting, be set quite high. I have it set to the max of 25% which seems to still be triggered by real motion, but not by general lighting changes.

ccrisan, 2015-08-07:
Thanks for sharing your findings and your configuration.

SuperSharpShooter, 2015-08-07:
25% noise level was causing it to miss some real motion, so I've found that 12% noise level, and 2.5% frame change threshold is giving more reliable results so far. Seems like each situation will require slight tuning of these values.

ccrisan, 2015-08-09:
The next version will use motion-mmal by default for CSI cameras.