ccrisan/motionpie #248: rest api ?

caguas, 2015-07-22:
Is it possible to have a rest api so devices such as iPhone can control the raspberry pi camera?

ccrisan, 2015-07-22:
Having a REST API would be nice indeed for various use cases. Unfortunately there's no such API available yet.

caguas, 2015-07-22:
Hi,

I modified ffmpeg to take advantage of the raspberry’s GPU to encode
h264 (hardware assist h264 encoding) . Would you be intested in getting
a copy of my version of ffmpeg. The changes are not too extensive…

My interest is to have a nice programmable network camera that is
cheap.

Amancio

ccrisan, 2015-07-22:
I don't see how this is related to a RESTful API, but yes, of course I am interested in your ffmpeg changes. As a matter of fact, I have planned support for h264 encoding for some time and I just need a weekend or two to actually implement it. Any tips regarding h264 are welcomed.

caguas, 2015-07-22:
Hi,

H264 hardware assist is not related to a RESTful api so I will open up another ‘issue’ or ‘ticket’ and
will upload my version of ffmpeg later on today. 

Cheers
Amancio

ccrisan, 2015-07-22:
Yes, please do. Thanks.

Ringerl, 2015-12-15:
An API would be great, e.g. to start/stop the camera. I'm currently doing this by (un)binding the usb device. Any idea how to do this more elegantly till the API is ready?

caguas, 2015-12-19:
Perhaps  a start point  for implementing rest is:

TodoMVC <http://todomvc.com/>

———————————————————————————————

As a summary, I want to access the application
via a native iOS APP, Safari in my iPhone as well
as Safari in my macbook. Additionally, I want to
be able to have a choice of sending the stream 
to a local device, a cloud storage such as dropbox,
or to my iPhone  and of course the ability to
remotely watch the video feed.

With respect to architecture, Douglas C. Schmidt
has written extensively about it and has online 
courses about patterns  and frameworks
for implementing systems such as this
surveillance appliance:

Digital Learning Offerings <http://www.dre.vanderbilt.edu/~schmidt/DigitalLearning/>

—————————————————————————————

Currently, I am evaluating Backbone.js and Underscore.js
to help me implement an application with a rest api.

Enjoy!
Amancio