ccrisan/motionpie #250: hardware assist h264 encoding for pi2

caguas, 2015-07-24:
These are modifications to motion and ffmpeg to
perform hardware assist h264 encoding on
raspberry pi 2 .

ffmpeg-2.6.1/libavcodec/vc264.c is the video encoder
that performs hardware assist encoding for the RPI2.

the command used to configure ffmpeg :
./configure --enable-gpl --disable-decoder=opus --enable-encoder=vc264 --enabl
e-nonfree --extra-libs=-ldl --extra-cflags=' -pipe '

motion/ffmpeg.c was modified to recognize the h264 extension

To see how to do hardware assist h264 encoding  please reference
encode.c which was provided by Broadcam. The included
file was take from:
/opt/vc/src/hello_pi/hello_encode/encode.c

Amancio Hasty

https://www.dropbox.com/s/2smfsqauhe0wo5v/pi2h264.tar.gz?dl=0

ccrisan, 2015-07-24:
Thanks @caguas. I believe this optimization could help h264 encoding a lot. I will make this task one of my priorities.

Jasimo, 2015-07-25:
Hi could someone explain tis advantage to me? Smoother video recordings in motion?
I am asking because i wonder if it could be possible to record the video after motion is detecting something in raw h.264 and converting it afterwards to e.g. mp4 like in the project of http://elinux.org/RPi-Cam-Web-Interface

caguas, 2015-08-22:
Hi Calin,

Had you had time to evaluate my patches and do they work for you?
Amancio

ccrisan, 2015-08-24:
I haven't had the time to look over it yet. I will let you know as soon as I get my hands on this issue.

donjakobo, 2015-08-31:
How about H264 for rpi 1 b? Is it not supported? I was under the impression H264 was supported via hardware across all rpi's

caguas, 2015-08-31:
Hi Jakub,

My patches should work with the rpi 1b. How about downloading
the patches that I posted and let us know if it works or not
in your system?

Amancio

gururise, 2015-08-31:
What version of motion do you include in the archive file?  Is it mrdave branch with rtsp support?  I'm currently compiling the included ffmpeg + your patches on my Rpi2 right now.

caguas, 2015-08-31:
Hi ,

I think that I downloaded the current version and it appears
to have support for rtsp. I read CHANGELOG and it did not
state the version number at the top of the file...
Amancio

gururise, 2015-09-01:
Sorry, I wasn't able to compile your version of ffmpeg on the latest Raspbian (Raspberry Pi 2):
libavcodec/vc1dsp.c: In function ‘put_vc1_mspel_mc12_c’:
libavcodec/vc1dsp.c:731:1: internal compiler error: Segmentation fault
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.6/README.Bugs> for instructions.
The bug is not reproducible, so it is likely a hardware or OS problem.
common.mak:57: recipe for target 'libavcodec/vc1dsp.o' failed
make: *** [libavcodec/vc1dsp.o] Error 1

gururise, 2015-09-01:
Sorry, I wasn't able to compile your version of ffmpeg on the latest Raspbian (Raspberry Pi 2):
> ./configure --enable-gpl --disable-decoder=opus --enable-encoder=vc264 --enable-nonfree --extra-libs=-ldl --extra-cflags=' -pipe '
> make

LD	ffmpeg_g
/usr/bin/ld: cannot find -lilclient
/usr/bin/ld: libavformat/libavformat.a(movenc.o): bad reloc symbol index (0xe1a09b >= 0x105) for offset 0x25f2 in section `.debug_info'
libavformat/libavformat.a: error adding symbols: Bad value
collect2: ld returned 1 exit status
Makefile:122: recipe for target 'ffmpeg_g' failed
make: *** [ffmpeg_g] Error 1

caguas, 2015-09-01:
Hi,

Can you try recompiling in a different SD card. Over here I had
a couple of bad SD cards and I just got rid of them. One of them
was brand new 8(

Amancio

gururise, 2015-09-01:
The error has something to do with the ilclient library in Raspberry Pi.  (See: https://jan.newmarch.name/RPi/OpenMAX/ILClient/ ).

So if I build the ilclient library (as described in the above link), and symlink the libilclient.a file to /usr/local/lib/libilclient.a then the build process continues, but ends up hitting other errors (related to the ilclient library):

LD      ffmpeg_g
libavcodec/libavcodec.a(vc264.o): In function `vc264_close':
/home/gene/Downloads/pi2_h264_hw/ffmpeg-2.6.1/libavcodec/vc264.c:236: undefined reference to `ilclient_disable_port_buffers'
/home/gene/Downloads/pi2_h264_hw/ffmpeg-2.6.1/libavcodec/vc264.c:237: undefined reference to `ilclient_disable_port_buffers'
/home/gene/Downloads/pi2_h264_hw/ffmpeg-2.6.1/libavcodec/vc264.c:239: undefined reference to `ilclient_state_transition'
/home/gene/Downloads/pi2_h264_hw/ffmpeg-2.6.1/libavcodec/vc264.c:240: undefined reference to `ilclient_state_transition'
/home/gene/Downloads/pi2_h264_hw/ffmpeg-2.6.1/libavcodec/vc264.c:242: undefined reference to `ilclient_cleanup_components'
/home/gene/Downloads/pi2_h264_hw/ffmpeg-2.6.1/libavcodec/vc264.c:246: undefined reference to `ilclient_destroy'

caguas, 2015-09-01:
Hi,
Make sure that the  ilclient library comes after libavcodec.a 

c++ - Why does the order in which libraries are linked sometimes cause errors in GCC? - Stack Overflow <http://stackoverflow.com/questions/45135/why-does-the-order-in-which-libraries-are-linked-sometimes-cause-errors-in-gcc>

Cheers
Amancio

gururise, 2015-09-01:
Hi,
I downloaded the tar.gz that you provided in the DropBox link. So I need to modify the Makefile that you provided?  I'm still not sure where I need to make the modification, as I don't really see any place where -lilclient is coming before -lavcodec.

Searching for all instances of lilclient give the following:
```
$ grep -rnw . -e "lilclient"
./configure:5595:       prepend LDFLAGS ' -L/opt/vc/lib -lilclient  -lGLESv2 -lEGL -lopenmaxil -lbcm_host -lvcos -lvchiq_arm -lrt'
./config.mak:55:LDFLAGS= -L/opt/vc/lib -lilclient  -lGLESv2 -lEGL -lopenmaxil -lbcm_host -lvcos -lvchiq_arm -lrt   -Wl,--as-needed -Wl,-z,noexecstack -Wl,--warn-common -Wl,-rpath-link=libpostproc:libswresample:libswscale:libavfilter:libavdevice:libavformat:libavcodec:libavutil:libavresample
```

searching for lavcodec:
```
$ grep -rnw . -e "lavcodec"
./libavcodec/libavcodec.pc:12:Libs: -L${libdir}  -lavcodec -lm -lz -pthread -lrt -ldl
./doc/examples/pc-uninstalled/libavcodec.pc:11:Libs: -L${libdir} -Wl,-rpath,${libdir} -lavcodec -lm -lz -pthread -lrt -ldl
```

gururise, 2015-09-03:
Any other suggestions on how to get this to build on a Raspberry Pi 2?  I'd love to try it out.

caguas, 2015-11-30:
Hi,

I did a few more modifications and managed to fix the issue of the library not linking in.

https://www.dropbox.com/s/wg7yub3ehilncwk/pi-h264-motion.tar.gz?dl=0

These modifications allow motion/ffmpeg for  the
Raspberry Pi 2  to do hardware assist h264 encoding.
The benefit is that it  uses less of the CPU to do
h264 encoding and it can scale in terms of video resolution .
For example, motion/ffmpeg can now do 1280x720 video
encoding with ease using the Raspberry Pi 2.

The ffmpeg/libavcodec/vc264.c is the hardware assist
h264 encoder for ffmpeg. The encoder uses the ilclient library
interface to talk to the Videocore IV GPU.
The h264  ilclient interface was copied  from :
/opt/vc/src/hello_pi/hello_encode/encode.c

motion/ffmpeg seems to drop frames at 30 frames
per second. The resulting video from ffmpeg
is usable. I am not sure what is the source of the
problem;however, h264 hardware assist encoding via
raspivid does not seem to have this problem. Perhaps
the latest version of motion/ffmpeg does not have
this problem.

The latest release of ffmpeg has Videocore IV assist
h264 decoder using the mmal interface. At some point
it will be nice to use a single interface for accessing
the h264 functionality in the Videocore IV GPU.

You can verify that the hardware assist h264 encoding
is working by running raspivid which uses the mmal interface.

In summary with respect to hardware assist h264 processing
for the Raspberry Pi:
1. ffmpeg/libavcodec/vc264.c -- ilclient based h264 encoding
2. the latest version of ffmpeg has mmal based hardware assist
h264 decoding.
3. RaspiVid.c  -- mmal based h264 decoding
4. /opt/vc/src/hello_pi/hello_encode/encode.c ilclient based h264
encoding which I used for vc264.c

So we now have mmal and ilclient interfaces to access the GPU...

This motion package was tested with Raspberry Pi OS:
2015-09-24-raspbian-jessie.img.

On  a newly installed system you will have
to run raspi-config to expand the filesystem
and to configure the system.

In addition to configuring the system , these
are the steps that I took to get the latest
updates for the system. Somehow my
USB wifi does not work that well with
the current system so I have tested
the setup with an ethernet connection.

sudo apt-get update
sudo apt-get upgrade
sudo rpi-update
add to /etc/modules
bcm2835-v4l2
sudo apt-get install libjpeg9-dev

tar -xzf h264-working-2.tar.gz

cd eye/opt
sudo mv /opt/vc/src/hello_pi/Makefile.include /opt/vc/src/hello_pi/Makefile.include.orig
sudo cp Makefile.include /opt/vc/src/hello_pi/Makefile.include
sudo mv /opt/vc/src/hello_pi/libs/ilclient/Makefile /opt/vc/src/hello_pi/libs/ilclient/Makefile.orig
sudo cp Makefile /opt/vc/src/hello_pi/libs/ilclient/.
cd /opt/vc/src/hello_pi/libs/ilclient/
make
sudo cp libilclient.so /opt/vc/lib/.
sudo ldconfig

cd eye/ffmpeg
make >& build.txt

sudo make install

sudo eye/motion
make
sudo make install

edit motion.conf to configure motion.
sudo cp motion.conf /usr/local/etc/.
start motion in the foreforeground to verify
that is working:
motion -n

Enjoy
Amancio Hasty

ccrisan, 2015-12-01:
There certainly are a lot of improvements that can be brought to the PI when it comes to encoding and others. I have planned a few changes to the video formats and video encoding in general (not only for the PI) and I'll make sure to take your suggestion into consideration. Thanks for putting together all these instructions for us.

XFer012, 2016-01-16:
@Amancio Hasty:

if I understand well, you patched ffmpeg sources in order to add Videocore-based hardware acceleration for H.264 encoding.
 
Which source files did you patch?
Would it be possible to have instructions to modify the latest ffmpeg sources as well?

Thanks a lot, having hardware-accelerated ffmpeg on the RPi would be a dream come true for me!!

Fernando

caguas, 2016-01-16:
Ola Fernando,

From my readme.txt file included with my patch:
"The ffmpeg/libavcodec/vc264.c is the hardware assist
h264 encoder for ffmpeg. The encoder uses the ilclient library
interface to talk to the Videocore IV GPU.
The h264  ilclient interface was copied  from :
/opt/vc/src/hello_pi/hello_encode/encode.c
….

In summary with respect to hardware assist h264 processing
for the Raspberry Pi:
1. ffmpeg/libavcodec/vc264.c -- ilclient based h264 encoding
2. the latest version of ffmpeg has mmal based hardware assist
h264 decoding.
3. RaspiVid.c  -- mmal based h264 decoding
4. /opt/vc/src/hello_pi/hello_encode/encode.c ilclient based h264
encoding which I used for vc264.c”
—————————————————————————————————————————

I am going to submit the patches to ffmpeg.org <http://ffmpeg.org/> shortly and
it is not too much work to upgrade them to the latest
ffmpeg release. The issue there is in deciding which
interface to use mmal or ilclient for hardware assist
encoding and decoding of h264 ...

Cheers,
Amancio

XFer012, 2016-01-16:
Hello Amancio,

>From my readme.txt file included with my patch:

Yes, I did read the readme; I was trying to figure which mods you did to the ffmpeg sources, in order to apply them to the latest source tree. :-)

You did a great, GREAT work, IMHO yours is a huge service to the RPi community!
Can't wait to build and test your patched ffmpeg on my PIs. :-)

>I am going to submit the patches to ffmpeg.org <http://ffmpeg.org/> shortly and
>it is not too much work to upgrade them to the latest
>ffmpeg release. The issue there is in deciding which
>interface to use mmal or ilclient for hardware assist
>encoding and decoding of h264 ...

Excellent news.
I sure hope they make up their minds in a short time (about which interface to adopt). GPU-accelerated H.264 encoding (not only with Raspivid, that is) is, well, fantastic!!

Thanks A LOT for your work!

All the best,

Fernando

caguas, 2016-01-16:
Hello Fernando,

Fine, see if you can look up how to implement a codec for ffmpeg.
This way you will be able to learn more than just implementing 
h264 hardware assist for ffmpeg.

In the meantime , I will post patches on Monday or so for the
latest ffmpeg to both this list and the  ffmpeg group. 

Cheers,
Amancio

caguas, 2016-01-20:
Hi Fernando,

There is another tool that you can use to do h264 hardware encoding:

I installed gstreamer: 
sudo apt-get install  gstreamer1.0-tools

gst-launch-1.0 -vv v4l2src !  video/x-raw,format=I420, framerate=30/1, width=640, height=480  !    omxh264enc  target-bitrate=2000000 control-rate=variable ! filesink location=webcam.h264

Now using /opt/vc/src/hello_pi/hello_video/hello_video.bin
hello_video.bin webcam.h264 

Uses hardware accelerated video playback.

Prior to compiling hello_video.bin you will have to compile and install:
/opt/vc/src/hello_pi/libs/ilclient 

There is lots of info on gstreamer on the net…

Enjoy
Amancio

XFer012, 2016-01-20:
Hello Amancio,

I'm encountering a few issues.

I'm trying to use the hardware-accelerated version of ffmpeg to encode the video stream from a USB webcam in H.264. 
I've compiled and installed ilclient and compiled ffmpeg without issues, but I can't use the hardware-accelerated encoder, because the "bcm2835-v4l2" kernel module refuses to load (Raspberry Pi 2 here).

Apparently, the module won't load if the RaspiCamera is not present (I have a generic USB webcam, correctly recognized and supported).

Of course that would defeat the purpose of having accelerated ffmpeg: if I had the RaspiCam, I could use RaspiVid which is already hardware-accelerated vs. H.264. :-(

Any hint?

Thanks!

Fernando

PS: Is there some other forum you partecipate to? I'm ashamed, I hijacked MotionPie's github comment system! :-)

caguas, 2016-01-20:
Hi,
Try the raspberry pi forums:

Raspberry Pi • View topic - USB device not working? Click here! <https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=53832>

ffmpeg opens /dev/video0 which should map to your usb camera.

Have Fun!
Amancio

XFer012, 2016-01-20:
Sorry Amancio, I think you did not read my message thoroughly.

The problem is not the USB camera.
The problem is that the "bcm2835-v4l2" kernel module, needed for OpenMax hardware acceleration, is NOT loaded if the Raspberry Camera Module is not mounted on the board.

So, no Hardware-accelerated ffmpeg available for a Raspberry without the Camera Module.

caguas, 2016-01-20:
The acceleration on ffmpeg is done via ilclient and I don’t think that I am
using anything  about the PI Camera . The accelerated module on
ffmpeg is given an array of pixels captured by the camera and
the module  has no idea where the pixels come from.

Later on, I will take my RPI apart and see if I can duplicate the problem…

In the mean time , may be you can try gstreamer. It takes about 5 minutes
to install and test…

Cheers
Amancio

caguas, 2016-01-20:
I just tried h264  hardware encoding without the PI Camera installed and
it seems to work:

/opt/vc/src/hello_pi/hello_encode $ ./hello_encode.bin fun.h264
the output was displayed on the screen with:

/opt/vc/src/hello_pi/hello_video $ ./hello_video.bin fun.h264

which is hardware accelerated and both programs use ilclient…

Cheers
Amancio

XFer012, 2016-01-20:
Unfortunately I can't try gstreamer: it needs 250MB+ of storage space because of a plethora of (useless) dependencies on Raspbian.

Will post a log of the error "your" ffmpeg gives about not having the bcm2835-v4l2 kernel module loaded, in the meantime MUCH THANKS for your work and support!

May I find you somewhere else? We're hijacking MotionPie space... :-)

caguas, 2016-01-20:
Sorry I can’t help you . Try the raspberry pi usb forum . Perhaps others
over there are having the same problem…

Good Luck
Amancio

caguas, 2016-01-22:
I tried accelerated ffmpeg without the pi camera and it seems to work over here.
also I seem to recollect to using openmax without a pi camera and everything
seem to work. 
Amancio

----------------------------------------
fmpeg -f rawvideo -vcodec rawvideo  -s 640x480 -r 25 -pix_fmt yuv420p  -i test.yuv  outputfile.h264
ffmpeg version 2.8.git Copyright (c) 2000-2016 the FFmpeg developers
  built with gcc 4.9.2 (Raspbian 4.9.2-10)
  configuration: --enable-gpl --disable-decoder=opus --enable-encoder=vc264 --enable-nonfree --extra-libs=-ldl --extra-cflags=' -pipe '
  libavutil      55. 13.100 / 55. 13.100
  libavcodec     57. 22.100 / 57. 22.100
  libavformat    57. 21.101 / 57. 21.101
  libavdevice    57.  0.100 / 57.  0.100
  libavfilter     6. 25.100 /  6. 25.100
  libswscale      4.  0.100 /  4.  0.100
  libswresample   2.  0.101 /  2.  0.101
  libpostproc    54.  0.100 / 54.  0.100
[rawvideo @ 0x252f2b0] Estimating duration from bitrate, this may be inaccurate
Input #0, rawvideo, from 'test.yuv':
  Duration: 00:00:00.04, start: 0.000000, bitrate: 92160 kb/s
    Stream #0:0: Video: rawvideo (I420 / 0x30323449), yuv420p, 640x480, 92160 kb/s, 25 tbr, 25 tbn, 25 tbc
Output #0, h264, to 'outputfile.h264':
  Metadata:
    encoder         : Lavf57.21.101
    Stream #0:0: Video: h264 (vc264), yuv420p, 640x480, q=-1--1, 25 fps, 25 tbn, 25 tbc
    Metadata:
      encoder         : Lavc57.22.100 vc264
Stream mapping:
  Stream #0:0 -> #0:0 (rawvideo (native) -> h264 (vc264))
Press [q] to stop, [?] for help
frame=    1 fps=0.0 q=-0.0 Lsize=       0kB time=00:00:00.04 bitrate=   3.6kbits/s speed=6.57x

dweebler2k, 2016-02-08:
Hi Armancio -
First thank you for the hard work, I am very excited to test this code on the pi.  Can post a working disk image?  I have tried to compile using your instructions several times and could not get it to complete.  A disk image may be the easiest option.  Thank you again!

GabrielDelepine, 2016-03-08:
@dweebler2k : Could you compile ffmpeg for the pi using this patch ? I am looking for a pre-compiled exec to test it rapidly. Thanks.

Martini-man, 2016-03-09:
Hi!

Can you make this for Raspberry PI 3?

Thanks.

ccrisan, 2016-03-09:
Check out the latest motioneyeos prerelease.

closed 2016-03-09 by ccrisan

blast1987, 2016-03-14:
Hello, just a question, how can i build only ffmpeg for pi 3 from latest motioneyeos prerelease to use it on raspbian jessie ?

dweebler2k, 2016-03-14:
The official instructions are here:
https://trac.ffmpeg.org/wiki/CompilationGuide/RaspberryPi

But I prefer these:
http://www.jeffreythompson.org/blog/2014/11/13/installing-ffmpeg-for-raspberry-pi/

blast1987, 2016-03-14:
Will it have hw assist if i build using this guide? 
I've already built for my pi2 some times ago using this: https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=53936

dweebler2k, 2016-03-14:
No it will not have hardware assist.  Amancio and Dickon seem to have the only working versions of hardware assist but I have been unable to reproduce working versions of either.  Dickon's version (https://github.com/dickontoo/omxmotion) fails to compile because of toolchain issues and Amancio's (above) compiles but I get only black frames.  

Best of luck and if you are able to get a hardware assist working please share.

dweebler2k, 2016-03-14:
Hi Gabriel -
I'd be happy to try.  Which patch were you referring to?

Martini-man, 2016-03-14:
I am also very interested in hw assist for pi 3.

XFer012, 2016-03-14:
"Amancio's (above) compiles but I get only black frames. "

I tried it, too. I get strange multicolored frames (but at least the underlying image is recognizeable) up to 352x288. At higher resolutions, I only get segfaults. :(

caguas, 2016-03-14:
Hi,

Which jessie version are you using and raspberry version ? 

You can invoke “motion” in the foreground with gdb 
and if the program crashes show us the stack trace.

Will recompile my patches with latest jessie later on and
will test it with my raspberry pi 2 and raspberry pi 3.

Amancio

dweebler2k, 2016-03-14:
Hi Amancio -
Any chance you could post the complete disk image?

It seems that would be the easiest starting point. Once we have a working version we can all attempt to tweak it together. Thanks!

XFer012, 2016-03-15:
We are hijacking motionpie's comment system. Maybe it would be better to discuss ffmpeg RPi hw assist somewhere else? Just asking.
Anywhere, I digged through Amancio's code a bit. A few comments:
- bitrate is fixed at 1 mbit/s. I changed the code to honour caller's bitrate value
- video frame format is 4:2:0 packed. Usually a webcam raw stream is in 4:2:2 format. This may, or may not, be related to my "strange colors" issue. Unfortunately, while 4:2:2 is listed among OMX formats (see .h), it results in an error 

In general, it's very important to follow Amancio's build instructions *to the letter*. Expecially the "fooling around" with ILClient library Makefiles. Otherwise the modified ffmpeg won't compile.

@blast1987 : the patched ffmpeg source tree is under eye/ffmpeg. Just issue "make -j4" there.

blast1987, 2016-03-15:
@XFer012 yes i think the same, maybe we should discuss somewhere else..

anyway compiling eye/ffmpeg will output a ffmpeg version that works with hardware for transcoding media files (i'm on pi3 now)?

XFer012, 2016-03-15:
In theory, yes (specifying "-vcodec vc264" instead of i.e. "-vodec libx264" among ffmpeg parameters).
In practice, as said, I did not manage to get a properly encoded video. :(

blast1987, 2016-03-15:
I'll try later and report back, 
should we open a topic about this on raspberrypi.org forum?

caguas, 2016-03-15:
Lets vote on where to take this topic:
1. raspberry pi forum
2.  ffmpeg

At any rate, my readme detailed enough info to get 
the ball rolling for hardware assist h264 .
Don’t forget that /opt/vc/src/hello_pi/hello_encode/encode.c
does hardware assist h264 encoding and my 
ffmpeg driver is based on this code.

XFer012, 2016-03-15:
I vote for RPi forum

blast1987, 2016-03-15:
I vote rpi forum

dweebler2k, 2016-03-16:
Hi Amancio -

Any chance you can post a working disk image?  I was able to compile but the resulting video was all black frames unless I used a raspicam at which the video encoded but did not look proper (very distorted). It would be great to test from a working build first. 

Thoughts?

dweebler2k, 2016-03-16:
Any forum works. Please let me know which you choose. Thanks!

caguas, 2016-03-16:
Hi,

I have decided to start the “thread” in the rpi forum. When I 
am ready to start the thread I will post the note to move
from this forum to the rpi forum.

In the mean time while I wrap up what I am doing in here,
people can compile  /opt/vc/src/hello_pi/hello_encode/encode.c
and see if it works in their system. If encode.c does not
work in your system just post in the rpi forum.

The reason for this approach is to use a very simple example of 
the code that we are using to see if it works in your system.

Prior to me modifying ffmpeg I use “encode.c" to find out if 
the underlying hardware was working properly. There is
a high probability that if encode.c does not work in your
system that you probably have something wrong with 
your system.

Have Fun!
Amancio

XFer012, 2016-03-16:
> people can compile /opt/vc/src/hello_pi/hello_encode/encode.c and see if it works in their system

I recompiled everything under /opt/vc (using rebuild.sh), and encode.c works fine on my Raspberry (the resulting video file plays without glitches).

But vc264.c does not work: the video is there, you can discern the recorded subject, but colors are totally messed up (upper part of the frame is BW only, mid part is strangely colored and bottom part has flashing squares).

Plus, it only "works" (not really, as described) only up to 320x240. If I try a higher resolution, it segfaults.

Of course everything works fine using software encoding (libx264).

dweebler2k, 2016-03-16:
I got the exact same results which is why I'm requesting a working disk image to start from. 

It would be great to know how to build from scratch but with a working instance we can all refine and benchmark.

XFer012, 2016-03-16:
Attaching the two videos, to show what I mean.

"hello_encode.mp4" is the output from encode.c under /opt/vc/...
"out_vc264.flv" is the output from ffmpeg using -vcodec vc264

[videos.zip](https://github.com/ccrisan/motionpie/files/177016/videos.zip)

caguas, 2016-03-16:
Hi Guys,

I will build again everything from scratch over here with the latest jessie.
When I do this I start with a fresh install of the latest jessie and this 
takes a little while . So tomorrow I will post on RPi to start the 
thread. 

motion works with no problems over here and I have tested
at 640x480.

These are the steps that I used to build the system from scratch:

sudo apt-get update
sudo apt-get upgrade
sudo rpi-update
add to /etc/modules
bcm2835-v4l2
sudo apt-get install libjpeg9-dev

tar -xzf h264-working-2.tar.gz

cd eye/opt
sudo mv /opt/vc/src/hello_pi/Makefile.include /opt/vc/src/hello_pi/Makefile.include.orig
sudo cp Makefile.include /opt/vc/src/hello_pi/Makefile.include
sudo mv /opt/vc/src/hello_pi/libs/ilclient/Makefile /opt/vc/src/hello_pi/libs/ilclient/Makefile.orig
sudo cp Makefile /opt/vc/src/hello_pi/libs/ilclient/.
cd /opt/vc/src/hello_pi/libs/ilclient/
make
sudo cp libilclient.so /opt/vc/lib/.
sudo ldconfig

cd eye/ffmpeg
make >& build.txt

sudo make install

cd eye/motion
make
sudo make install

edit motion.conf to configure motion.
sudo cp motion.conf /usr/local/etc/.

start motion in the foreforeground to verify that is working

   motion -n

**Remember that vc264.c is based on encode.c **

Cheers
Amancio

XFer012, 2016-03-16:
Amancio,

yes, you posted the instructions many times, and we followed them to the letter.
And yes, it's absolutely clear that vc264.c is based upon encode.c. 
Still, we could not get it to produce a useble H.264 video with ffmpeg.

We are trying to use ffmpeg with vc264 to encode video, not only to work with motion.

Our purpose is to get a properly encoded H.264 video, with right colors etc.

Thanks for your support.

Fernando

Martini-man, 2016-03-17:
He said he will do it tomorrow with the latest jessie and hopefully with RPI3.

He will run into the same problems as we do and give us the solution.

XFer012, 2016-03-19:
News? Let's open the thread on raspberry forum, at least...?

[9 more replies]