ccrisan/motionpie #251: WiFi: Unable to setup wifi connection

kevinp79, 2015-07-24:
I am using the latest version for Raspberry Pi 2, as of July 24th, 2015.  My network is a straightforward WPA2 configuration. 

I am able to see the web setup page while the LAN cable is plugged in.  After configuring the wifi settings for DHCP and entering my SSID and password, all seems to be okay.  I can even unplug the LAN cable and the connection is maintained on wifi.  However, after rebooting the device with only the wifi dongle I will never be able to connect again solely on wifi.

I have tried multiple clean SD card images with the same result.  The dongle is a common Edmax.

Is there a certain order of installation procedures that I should try?  Should I have the dongle installed from the very beginning?

Would love to see this work.
Thanks

ccrisan, 2015-07-24:
Could you please send me your logs? See [Reporting An Issue](https://github.com/ccrisan/motionpie/wiki/Reporting-An-Issue#attach-log-files) if you don't know how.

kevinp79, 2015-07-25:
This was my error on my DHCP setup.  I neglected to realize that the MAC address changed when I switched to the wifi dongle.  I was accessing the IP for the LAN NIC on the Pi.  Oops.

Thanks for your time.

closed 2015-07-25 by kevinp79