ccrisan/motionpie #253: building fails

ebiancarelli, 2015-07-26:
When building from the current git repo, I get a failure to build python-pycurl. 
```
Could not run curl-config: [Errno 2] No such file or directory
```

Any hints to get past this?

```
 ~/motionpie$ ./build.sh raspberrypi
>>> python-pycurl 7.19.5.1 Installing to target
(cd /home/ernie/motionpie/output/raspberrypi/build/python-pycurl-7.19.5.1//; PATH="/home/ernie/motionpie/output/raspberrypi/host/bin:/home/ernie/motionpie/output/raspberrypi/host/sbin:/home/ernie/motionpie/output/raspberrypi/host/usr/bin:/home/ernie/motionpie/output/raspberrypi/host/usr/sbin:/home/ernie/bin:/opt/local/bin:/opt/local/sbin:/usr/local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games" CC="/home/ernie/motionpie/output/raspberrypi/host/usr/bin/ccache /home/ernie/motionpie/output/raspberrypi/host/usr/bin/arm-buildroot-linux-gnueabihf-gcc" CFLAGS="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -pipe -mhard-float -O2 " LDFLAGS="" LDSHARED="/home/ernie/motionpie/output/raspberrypi/host/usr/bin/arm-buildroot-linux-gnueabihf-gcc -shared" PYTHONPATH="/home/ernie/motionpie/output/raspberrypi/target/usr/lib/python2.7/sysconfigdata/:/home/ernie/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/" _python_sysroot=/home/ernie/motionpie/output/raspberrypi/host/usr/arm-buildroot-linux-gnueabihf/sysroot _python_prefix=/usr _python_exec_prefix=/usr  /home/ernie/motionpie/output/raspberrypi/host/usr/bin/python setup.py install --prefix=/home/ernie/motionpie/output/raspberrypi/target/usr )
Traceback (most recent call last):
  File "setup.py", line 634, in <module>
    ext = get_extension(split_extension_source=split_extension_source)
  File "setup.py", line 392, in get_extension
    ext_config = ExtensionConfiguration()
  File "setup.py", line 65, in __init__
    self.configure()
  File "setup.py", line 100, in configure_unix
    raise ConfigurationError(msg)
__main__.ConfigurationError: Could not run curl-config: [Errno 2] No such file or directory
make: *** [/home/ernie/motionpie/output/raspberrypi/build/python-pycurl-7.19.5.1/.stamp_target_installed] Error 1
```

ebiancarelli, 2015-07-26:
[SOLVED] needed to install libcurl4-openssl-dev package on my build machine. too bad the buildroot version of libcurl can't be used.

closed 2015-07-26 by ebiancarelli

ccrisan, 2015-08-09:
The latest git version should work fine with the staging buildroot version of libcurl.