ccrisan/motionpie #254: Editing default values for V4L2 driver

MMUBigBear, 2015-07-28:
Good day to all!
First of all, I've got MotionPie up and running on my RPi2 and have been testing out its features. However, I can't seem to edit/change the default values for v4l2-ctl, namely the auto_exposure property. I am able to change the value from '0' to '1' using v42l-ctl --set-ctrl=auto_exposure=1, but it will be reset to '0'(default) every time the Pi reboots.

Any help is much appreciated, thank you!

ccrisan, 2015-07-28:
There's no friendly way to do that, but you could use the [userinit.sh](https://github.com/ccrisan/motionpie/wiki/Tweaks#adding-custom-startup-commands) script to preconfigure your camera at boot.

closed 2015-07-28 by ccrisan

MMUBigBear, 2015-07-29:
Thanks for the tip! Did that and now it's working like how I need it to.
P/S: Windows users can create the userinit.sh file by typing the command you like in notepad, eg
  v42l-ctl --set-ctrl=auto_exposure=1
save it as a .txt file, and then simply rename it userinit.sh
Transfer it to your Pi's /data/etc, reboot and voila!

Thanks Calin for the fantastic work on MotionPie!