ccrisan/motionpie #256: Out of space in memory card but not media visible

betocerda, 2015-07-30:
Disk usage is at 89% but there is no media visible if I click the media icon for any of my 3 cameras. Is there any place to look for that space occupied by junk files?

ccrisan, 2015-07-30:
Have you enabled debugging in Expert Settings? It's probably the /data/logs that fill up your SD card.

betocerda, 2015-07-31:
I cant find the option to enable/disable the logs. I'm able to open the log files and they are small files so they don't seem to be the issue. motioneye.log is the biggest one and it is only 1,532 KB. Any other ideas?

ccrisan, 2015-07-31:
Can you SSH into your Pi and run the `df` command as well as the `du -h /data` command?

dkelley810, 2015-08-01:
I am having the same issues. But I can not SSH to my Pi. I can get into the GUI. I can not ssh. I sent you the files in email.

ccrisan, 2015-08-01:
What do you mean you cannot ssh? What error do you get?

ccrisan, 2015-08-01:
Can you at least connect a monitor and a keyboard and log in on the normal terminal to run the du commands?

dkelley810, 2015-08-01:
i was able to fix it on my end. I removed the sdcard and moved the images on my laptop.
it looks like i filled the sdcard up and it was not able to access it by ssh. Thank You,Dennis

ccrisan, 2015-08-01:
The question still stands for @betocerda.

betocerda, 2015-08-02:
df -h /data returns: 
4.0K /data/media

df returns
Filesystem                Size      Used Available Use% Mounted on
/dev/root               116.2M     96.0M     11.8M  89% /
devtmpfs                429.3M         0    429.3M   0% /dev
/dev/mmcblk0p1           16.0M      6.9M      9.0M  44% /boot
/dev/mmcblk0p3           29.0G     26.5G   1002.1M  96% /data
tmpfs                   433.4M         0    433.4M   0% /dev/shm
tmpfs                   433.4M      2.9M    430.5M   1% /tmp
/dev/mmcblk0p3           29.0G     26.5G   1002.1M  96% /home/ftp/sdcard
/dev/mmcblk0p3           29.0G     26.5G   1002.1M  96% /home/ftp/storage
[root@mp-251a82bf ~]# cd home
-sh: cd: home: No such file or directory
[root@mp-251a82bf ~]# cd..
-sh: cd..: command not found
[root@mp-251a82bf ~]# cd ..
[root@mp-251a82bf home]# ls
ftp/
[root@mp-251a82bf home]# cd ..
[root@mp-251a82bf /]# ls
System.map  dev/        lib32@      media/      programs/   sys/
bin/        etc/        libexec/    mnt/        root/       tmp/
boot/       home/       linuxrc@    opt/        run@        usr/
data/       lib/        lost+found/ proc/       sbin/       var/
[root@mp-251a82bf /]# de dev
-sh: de: command not found
[root@mp-251a82bf /]# ls
System.map  dev/        lib32@      media/      programs/   sys/
bin/        etc/        libexec/    mnt/        root/       tmp/
boot/       home/       linuxrc@    opt/        run@        usr/
data/       lib/        lost+found/ proc/       sbin/       var/
[root@mp-251a82bf /]# cd dev
[root@mp-251a82bf dev]# ls
autofs              ram14               tty23               tty53
block/              ram15               tty24               tty54
bus/                ram2                tty25               tty55
cachefiles          ram3                tty26               tty56
char/               ram4                tty27               tty57
console             ram5                tty28               tty58
cpu_dma_latency     ram6                tty29               tty59
disk/               ram7                tty3                tty6
fb0                 ram8                tty30               tty60
fd@                 ram9                tty31               tty61
full                random              tty32               tty62
hwrng               raw/                tty33               tty63
kmsg                shm/                tty34               tty7
log=                stderr@             tty35               tty8
media0              stdin@              tty36               tty9
mem                 stdout@             tty37               ttyAMA0
memory_bandwidth    tty                 tty38               ttyprintk
mmcblk0             tty0                tty39               urandom
mmcblk0p1           tty1                tty4                v4l/
mmcblk0p2           tty10               tty40               vc-cma
mmcblk0p3           tty11               tty41               vc-mem
network_latency     tty12               tty42               vchiq
network_throughput  tty13               tty43               vcio
null                tty14               tty44               vcs
ppp                 tty15               tty45               vcs1
ptmx                tty16               tty46               vcsa
pts/                tty17               tty47               vcsa1
ram0                tty18               tty48               vcsm
ram1                tty19               tty49               video0
ram10               tty2                tty5                watchdog
ram11               tty20               tty50               zero
ram12               tty21               tty51
ram13               tty22               tty52
[root@mp-251a82bf dev]# cd ..
[root@mp-251a82bf /]# ls
[... 69 lines cut ...]
-sh: /dev/mmcblk0p1: Permission denied
[root@mp-251a82bf sdcard]# /dev/mmcblk0p3           29.0G     26.5G    989.4M  96% /data
-sh: /dev/mmcblk0p3: Permission denied
[root@mp-251a82bf sdcard]# tmpfs                   433.4M         0    433.4M   0% /dev/shm
-sh: tmpfs: command not found
[root@mp-251a82bf sdcard]# tmpfs                   433.4M      2.9M    430.5M   1% /tmp
-sh: tmpfs: command not found
[root@mp-251a82bf sdcard]# /dev/mmcblk0p3           29.0G     26.5G    989.4M  96% /home/ftp/sdcard
-sh: /dev/mmcblk0p3: Permission denied
[root@mp-251a82bf sdcard]# /dev/mmcblk0p3           29.0G     26.5G    989.4M  96% /home/ftp/storage
-sh: /dev/mmcblk0p3: Permission denied
[root@mp-251a82bf sdcard]# clear
[root@mp-251a82bf sdcard]# df
Filesystem                Size      Used Available Use% Mounted on
/dev/root               116.2M     96.0M     11.8M  89% /
devtmpfs                429.3M         0    429.3M   0% /dev
/dev/mmcblk0p1           16.0M      6.9M      9.0M  44% /boot
/dev/mmcblk0p3           29.0G     26.5G    987.3M  96% /data
tmpfs                   433.4M         0    433.4M   0% /dev/shm
tmpfs                   433.4M      2.9M    430.5M   1% /tmp
/dev/mmcblk0p3           29.0G     26.5G    987.3M  96% /home/ftp/sdcard
/dev/mmcblk0p3           29.0G     26.5G    987.3M  96% /home/ftp/storage
[root@mp-251a82bf sdcard]#

I cannot see any files wit ls in /home/ftp/sdcard, the same for /home/ftp/storage

ccrisan, 2015-08-02:
I was more interested in du -h /data.

betocerda, 2015-08-02:
This is the result
[root@mp-251a82bf ~]# du -h /data
4.0K    /data/media

ccrisan, 2015-08-02:
I doubt that's the only output. Did you interrupt the process before it finished? You need to let it run and scan all the folders.

betocerda, 2015-08-02:
You are righm here is the full output

[root@mp-251a82bf ~]# du -h /data
4.0K    /data/media
4.0K    /data/etc/crontabs
4.0K    /data/etc/ssh
26.6G   /data/etc
4.0K    /data/log/samba/cores/nmbd
4.0K    /data/log/samba/cores/smbd
12.0K   /data/log/samba/cores
336.0K  /data/log/samba
2.3M    /data/log
16.0K   /data/lost+found
1.1G    /data/output/2015-08-02
1.2G    /data/output/2015-08-01
2.3G    /data/output
28.9G   /data
[root@mp-251a82bf ~]#

ccrisan, 2015-08-02:
Alright. Now what does "ls -alh /data/etc" say?

betocerda, 2015-08-02:
It is returning a huge list of jpg files. I have the Still Images option disabled.

Something like this:
Many jpg files before and then:
-rw-r--r--    1 root     root       24.6K Jul 11 09:20 99-20150711092039-01.jpg
-rw-r--r--    1 root     root       23.1K Jul 17 23:02 99-20150717230201-01.jpg
-rw-r--r--    1 root     root       24.0K Jul 20 08:50 99-20150720085025-01.jpg
-rw-r--r--    1 root     root       24.0K Jul 20 08:50 99-20150720085025-02.jpg
drwxr-xr-x    2 root     root        4.0K Dec 31  1969 crontabs/
-rw-r--r--    1 root     root          72 Jul  1 21:18 date.conf
lrwxrwxrwx    1 root     root          37 Jul  1 21:18 localtime -> /usr/share/zoneinfo/America/Monterrey
-rw-r--r--    1 root     root         239 Jul  1 21:18 motion.conf
-rw-r--r--    1 root     root          81 Jul  1 21:18 motioneye.conf
-rw-r--r--    1 root     root           0 Jul  1 21:18 proftpd.conf
-rw-------    1 root     root          53 Dec 31  1969 shadow
-rw-------    1 root     root          41 Dec 31  1969 shadow-
-rw-r--r--    1 root     root           0 Jul  1 21:18 smb.conf
drwxr-xr-x    2 root     root        4.0K Dec 31  1969 ssh/
-rw-------    1 root     root         668 Dec 31  1969 ssh_host_dsa_key
-rw-r--r--    1 root     root         606 Dec 31  1969 ssh_host_dsa_key.pub
-rw-------    1 root     root         227 Dec 31  1969 ssh_host_ecdsa_key
-rw-r--r--    1 root     root         178 Dec 31  1969 ssh_host_ecdsa_key.pub
-rw-------    1 root     root         411 Dec 31  1969 ssh_host_ed25519_key
-rw-r--r--    1 root     root          98 Dec 31  1969 ssh_host_ed25519_key.pub
-rw-------    1 root     root         981 Dec 31  1969 ssh_host_key
-rw-r--r--    1 root     root         646 Dec 31  1969 ssh_host_key.pub
-rw-------    1 root     root        1.6K Dec 31  1969 ssh_host_rsa_key
-rw-r--r--    1 root     root         398 Dec 31  1969 ssh_host_rsa_key.pub
-rw-r--r--    1 root     root          77 Jul  1 21:18 static_ip.conf
-rw-r--r--    1 root     root        1.2K Mar 26 22:40 thread-1.conf
-rw-r--r--    1 root     root        1.2K Mar 31 13:57 thread-2.conf
-rw-r--r--    1 root     root        1.2K Mar 24 20:16 thread-3.conf
-rw-r--r--    1 root     root           0 Jun 16 20:48 thread-4.conf
-rw-r--r--    1 root     root         171 Jul  1 21:18 watch.conf
-rw-r--r--    1 root     root         134 Jul  1 21:18 wpa_supplicant.conf

ccrisan, 2015-08-02:
I believe you somehow configured the root directory to empty or to something that points to /data/etc somehow. It then created all your media files there.

betocerda, 2015-08-02:
Last time I installed version 20150621 and then I loaded a previous configuration. Maybe that messed up something, I will try to install the most recent version and start with a new configuration.

ccrisan, 2015-08-03:
Yes, that is a possibility as there are various incompatibilities between configurations from different versions. I'll close this issue now as I think we have identified the problem and the solution is to clean up  the images in the /data/etc folder.

closed 2015-08-03 by ccrisan