ccrisan/motionpie #257: Another WiFi question

TR4Aman, 2015-07-31:
Brillian piece of sofware, exactly what i was looking for, thank you for the hard work
I see you have had this questions many times but not for a Tenda W522U or W133M wifi dongle.
I have MotionPie working fine when connected via ethernet but wirelessly with either wifi dongle i get the reboot loop. 
The contents of /data/etc/wpa_supplicant.conf is :
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
scan_ssid=1
psk="????????"
ssid="OAKDENE-PC_Network"
}

 /data/log/wpa_supplicant.conf is absent but i have wpa_supplicant.log

Successfully initilised wpa_supplicanty=
rfkill: cannot open RFKILL control device

I do have a staic_ip.conf in the data/etc folder but it is empty.

Ant suggestions would be a great help

many thanks

TR4Aman, 2015-07-31:
I should add when i boot up with the Tenda in the usb i get the following message on the screen

Ieee80211 phy0:rt2x00lib_request_firmware: info -Firmware detected –version:0.29
failed

ccrisan, 2015-07-31:
It looks like the device is not supported by the kernel. The kernel tries to load a firmware that doesn't seem to be the appropriate one. Does it work with raspbian?

TR4Aman, 2015-07-31:
Thanks for the rapid response . Yes it did work with raspian. 

I will try it again with a sd card with raspian on it

TR4Aman, 2015-07-31:
The Tenda W522U works perfrectly in Raspian

ccrisan, 2015-08-01:
Please send me the boot.log and the dmesg.log files.

TR4Aman, 2015-08-02:
When running MotionPie

 

 

However this is boot with Ethernet plugged in and no wifi dongle as I just get reboot cycle with dongle

If I reboot after having the dongle in with np Ethernet I get a different boot.log, but when I insert the Ethernet I cant access the file using ssh without a reboot.

What folder is the dmsg.log file in?

 

 

 

 

The boot.log

Setting hostname: done

Starting logging: done

Starting watchdog: done

Starting eudev: done

Setting up data partition

/dev/mmcblk0p3: Bad magic number in super-block while opening filesystem

Formatting data partition

Discarding device blocks:    4096/3802880^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H 528384/$

Creating filesystem with 3802880 4k blocks and 950976 inodes

Filesystem UUID: 32edf471-b0ce-4c6c-a0d5-e1dd9fcec2a5

Superblock backups stored on blocks:

        32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208

 

Allocating group tables:   0/117^H^H^H^H^H^H^H       ^H^H^H^H^H^H^Hdone        $

Writing inode tables:   0/117^H^H^H^H^H^H^H       ^H^H^H^H^H^H^Hdone           $

Creating journal (32768 blocks): done

Writing superblocks and filesystem accounting information:   0/117^H^H^H^H^H^H^$

ccrisan, 2015-08-03:
I understand the difficulty of this :) What you could do is to write the image from scratch to your SD card, boot with both ethernet and wifi adapter connected (but not configured!), and send me the dmesg that results from that boot. The error that I'm looking for will result from just connecting the adapter without actually using it.

TR4Aman, 2015-08-03:
Thanks for your patience with this matter.

I reimaged the SD card and rebooted with Wifi and Ethernet connected.

Here is a copy of the smesg file

 

Welcome to mp-1d2a8ef3!
admin@192.168.0.173's password:
[root@mp-1d2a8ef3 ~]# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 3.18.12-quick (ccrisan@supermicro) (gcc version 4.8                                                                                        .4 (Buildroot 2015.08-git-g73e1681) ) #1 PREEMPT Sun Jul 19 19:51:43 EEST 2015
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr                                                                                        =00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruc                                                                                        tion cache
[    0.000000] Machine: BCM2708
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 98304
[    0.000000] free_area_init_node: node 0, pgdat c0515b48, node_mem_map d7cf800                                                                                        0
[    0.000000]   Normal zone: 768 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 98304 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pag                                                                                        es: 97536
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1280                                                                                         bcm2708_fb.fbheight=1024 bcm2708.boardrev=0xe bcm2708.serial=0x1d2a8ef3 smsc95xx                                                                                        .macaddr=B8:27:EB:2A:8E:F3 bcm2708_fb.fbswap=1 sdhci-bcm2708.emmc_clock_freq=250                                                                                        000000 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.fiq_fix_en                                                                                        able=1 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait ro rootflags=no                                                                                        load
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 384156K/393216K available (4159K kernel code, 201K rwdata                                                                                        , 688K rodata, 128K init, 328K bss, 9060K reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
                   vmalloc : 0xd8800000 - 0xff000000   ( 616 MB)
                   lowmem  : 0xc0000000 - 0xd8000000   ( 384 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0xc0008000 - 0xc04c3f0c   (4848 kB)
                     .init : 0xc04c4000 - 0xc04e4000   ( 128 kB)
                     .data : 0xc04e4000 - 0xc0516580   ( 202 kB)
                      .bss : 0xc0516580 - 0xc0568824   ( 329 kB)
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:522
[    0.000022] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2                                                                                        147483648000ns
[    0.000067] Switching to timer-based delay loop, resolution 1000ns
[    0.000630] Console: colour dummy device 80x30
[    0.001495] console [tty1] enabled
[    0.001540] Calibrating delay loop (skipped), value calculated using timer fr                                                                                        equency.. 2.00 BogoMIPS (lpj=1000)
[    0.001603] pid_max: default: 32768 minimum: 301
[    0.001928] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.001984] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.002923] CPU: Testing write buffer coherency: ok
[    0.003381] Setting up static identity map for 0x415de8 - 0x415e20
[    0.005404] devtmpfs: initialized
[    0.022186] VFP support v0.3: implementor 41 architecture 1 part 20 variant b                                                                                         rev 5
[    0.023167] NET: Registered protocol family 16
[    0.028312] DMA: preallocated 4096 KiB pool for atomic coherent allocations
[    0.029308] bcm2708.uart_clock = 3000000
[    0.031014] mailbox: Broadcom VideoCore Mailbox driver
[    0.031164] bcm2708_vcio: mailbox at f200b880
[    0.031543] bcm_power: Broadcom power driver
[    0.031594] bcm_power_open() -> 0
[    0.031625] bcm_power_request(0, 8)
[    0.532317] bcm_mailbox_read -> 00000080, 0
[    0.532359] bcm_power_request -> 0
[    0.532556] Serial: AMBA PL011 UART driver
[    0.532742] dev:f1: ttyAMA0 at MMIO 0x20201000 (irq = 83, base_baud = 0) is a                                                                                         PL011 rev3
[    0.568674] usbcore: registered new interface driver usbfs
[... cut ...]
[    5.154428] hid-generic 0003:1241:1166.0003: input,hidraw2: USB HID v1.10 Mou                                                                                        se [HID 1241:1166] on usb-bcm2708_usb-1.3.3/input0
[   38.159566] EXT4-fs (mmcblk0p3): mounted filesystem with ordered data mode. O                                                                                        pts: (null)
[   39.619782] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   41.343990] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x45E

ccrisan, 2015-08-04:
I'm sorry but there's nothing suspicious in your log that would indicate any problem with the wifi adapter. If you say it works with Raspbian, then just hold on for a week or two until I manage to release a new version based on the latest kernel (4.0) used by Raspbian and let's hope for the best :)

TR4Aman, 2015-08-04:
OK, thanks for your help. I have to say the wifi dongle worked from the early days of Raspberry Pi so I don’t hold out much hope for the new Raspian kernal.  It worked on Sunday when I tried it with admittedly a non-updated version of Raspian.

However, I may have installed it manually as per the modifying the wpa-supplicant.conf file at the end of these instructions from Modmy pi

https://www.modmypi.com/blog/how-to-set-up-the-ralink-rt5370-wifi-dongle-on-raspian

 

I also not that in my dmesg file that I sent you, the id product for the usb wifi is 3572 that’s on line 3.199, but in the modmypi page it says it should RT5370.

 

If I don’t get it to work what would you suggest as a wifi dongle for motiopi

Regards

 

 

John

TR4Aman, 2015-08-04:
Sorry I was looking at and sending the wrong wifi dongle link

It this one and is as per the dmesg file ralink3572

 

http://www.modmypi.com/raspberry-pi/accessories/wifi-dongles/wifi-dongle-tenda-w522u-dual-band-5ghz

ccrisan, 2015-08-06:
Damn, this looks strange. I mean the firmware file should be right there (`/lib/firmware/rt3071.bin`), the module is probably loaded (`rt2800usb`) and the adapter appears to be fully supported under Linux. 

The only thing that I can do about your problem is to log in remotely (using the wired connection, of course), and try to manually connect and debug the wireless adapter. Would you be able to set up a remote login session (i.e. to forward port 22 or whatever to your motionPie, while it is connected wiredly)?

TR4Aman, 2015-08-06:
I have never done port forwarding before, but will give it a try, I will have a read about port forwarding on the various motionpie sites. Or is it very straight forward?

ccrisan, 2015-08-06:
It's a router thing and there's nothing motionPie-specific about it. If you give me your router model, I can look up the instructions online and send them to you.

TR4Aman, 2015-08-15:
Hi,

I have gone to my router advanced/port forwarding/ for my MotionPie 192.168.1 186 port 22.

If I look for my real ip address it comes up as 46.253.36.164

 

Was it as simple as that or I have underestimated the complexity of the task

 

 

Regards

 

John

ccrisan, 2015-08-16:
I cannot seem to connect to that IP address on port 22. Has your public IP address changed since? You may want to use a public port other than 22 (e.g. 2222) and forward it to 192...:22. This way you avoid attacks and your PI will be a lot happier.

TR4Aman, 2015-08-19:
Hi,

I have changed the port to 2222. I have TCP set to 2222 but nothing for UDP

The internal address is still 192.168.0.186:2222

 

46.253.36.164

Your public IP address

 

The Pi is switch on.

Let me know if that works

 

John

ccrisan, 2015-08-20:
I'm sorry but your public IP address doesn't seem to have port 2222 open.

TR4Aman, 2015-08-27:
I have now set mmotionpi ip static as suggested using your /data/etc/staic_ip.conf sample file

Using 192.168.0.101/24 I assume 24 is the port.

The Pi sets this static addres and all appears in the router status. I have changed the port forwarding for 192.168.9.101 to 24 

When all rebooted and login to motion pie I still get port 24 closed….

Is there anything else I can try

ccrisan, 2015-08-27:
/24 is the netmask (255.255.255.0, i.e. 24 bits). The incoming port could be anything (2222 is fine), just make sure to forward it to port 22.

TR4Aman, 2015-09-01:
Found out the issue with the port forward never opening. My IP here in Spain is via WiMax so it comes from a mast to an antenna on my roof. Anyway to cut a long storey short I need to somehow change the config of my router with this antenna then they will open a port from their end….or buy a router from them with the port configured.

Anyway might change provider

Thanks for all your help

 

John