ccrisan/motionpie #259: Wifi not working, working then restart looping

redfox05, 2015-08-02:
Hi there,
I'm using the recommended Edimax Wifi adapter, which works fine with raspbian.
First attempt using one of the June versions, it wasnt working for me, wifi just wouldnt work. Didnt know what diagnostics to try, so cant tell you that.

Then I updated today to 20150719, and the wifi works. However its very slow. And still, for stability, I would rather use a static ip rather than rely on DHCP, also nice to know which IP it's always on. I set this using the Static IP setting in the Web GUI. However, now the MotionPie seems to be stuck in a reboot loop. The light comes on for the camera LED, GUI works, SSH with root works, but then the light goes out, SSH is disconnected, and it seems to be restarting... Very odd. It's done this at least 3 times so far, and counting. Any suggestions?

redfox05, 2015-08-02:
Also having Permissions Denied issues when trying to create the static config file.

I found it odd that when I tried to set the static ip in Web GUI, the motionpie is now stuck in a restart loop, but while its running it still connects on the old DHCP IP rather than the static one I set.

ccrisan, 2015-08-02:
Please send me all the logs.

redfox05, 2015-08-02:
Hi, thanks for the quick reply. I'm working on it this afternoon. Which logs would you like?

redfox05, 2015-08-02:
Im actually considering wiping the SD all over and starting again.... :P

redfox05, 2015-08-02:
Problem. The camera LED is on, but its not registering on the network according to the router. It seems very unreliable right now. I wont be able to get you the logs, as I cant login without a network connection. I'll open up the PiCE and reformat the card maybe

ccrisan, 2015-08-02:
Do not reformat the card before recovering the logs. You can copy them from the sdcard directly, using a Linux PC. If you don't have one, I have no other solution for you.

redfox05, 2015-08-02:
sure, ill try a few hard restarts and see if it reconnects, else I'll copy the logs before the re-format. What are the locations of the log files you'd like?

ccrisan, 2015-08-02:
The entire /log directory would be interesting.

redfox05, 2015-08-02:
Hard restart worked, but then SSH froze... odd.

But I did manage to get a few commands in, cant find /log....

[root@mp-6ea4bff5 ~]# cd /log        
-sh: cd: /log: No such file or directory

ccrisan, 2015-08-02:
I thought you accessed the sdcard directly. The log directory is at /data/log.

redfox05, 2015-08-02:
ok, ill try again. I got kicked out with `[root@mp-6ea4bff5 ~]# Write failed: Broken pipe`

redfox05, 2015-08-02:
OK, the restarts are not down to the 1A adapter. Still getting `broken pipe` crashes in SSH.

boot.log
    ---- booting ----
    Setting hostname: done
    Starting logging: done
    Starting watchdog: done
    Starting eudev: done
    Setting up data partition
    Setting custom root/admin password
    Starting wpa_supplicant: failed
    ---- booting ----
    Setting hostname: done
    Starting logging: done
    Starting watchdog: done
    Starting eudev: done
    Setting up data partition
    Setting custom root/admin password
    Starting wpa_supplicant: failed
    ---- booting ----
    Setting hostname: done
    Starting logging: done
    Starting watchdog: done
    Starting eudev: done
    Setting up data partition
    Setting custom root/admin password
    Starting wpa_supplicant: failed

Cant get the other files. The pi crashes before I get a chance. I'll stick it into the laptop and get them that way. Back in 5 mins.

ccrisan, 2015-08-02:
Well, I'd like to see your dmesg.log and wpa_supplicant.log files.

redfox05, 2015-08-02:
Sure, I'll send those over. Any suggestions on mounting those partitions on linux/mac? I can't seem to get anything mounted over than the small partition with the config files.

ccrisan, 2015-08-02:
Well you should have something like /dev/mmcblk0p3 or something like the third partition on your sd card. Look for the /log directory in the root of that partition.

redfox05, 2015-08-02:
The Macs wont mount the ext3/4 partitions.

http://raspberrypi.stackexchange.com/questions/13695/edit-root-files-from-sd-card

I'll figure out another way though to get those logs

redfox05, 2015-08-02:
Plugging back into Ethernet, and removing the WiFi dongle. Lets see...
Wow, what a nightmare, finally found a way to stabilise the Pi, unplug the camera. Right, lets get those logs:

messages - not all, just latest
`

Jan  1 00:00:15 mp-6ea4bff5 daemon.err smbd[519]: [1970/01/01 00:00:15.699593,  0] ../source3/lib/util_sock.c:455(open_socket_in)
Jan  1 00:00:15 mp-6ea4bff5 daemon.err smbd[519]:   open_socket_in(): socket() call failed: Address family not supported by protocol
Jan  1 00:00:15 mp-6ea4bff5 daemon.err smbd[519]: [1970/01/01 00:00:15.700609,  0] ../source3/smbd/server.c:690(smbd_open_one_socket)
Jan  1 00:00:15 mp-6ea4bff5 daemon.err smbd[519]:   smbd_open_once_socket: open_socket_in: Address family not supported by protocol
Jan  1 00:00:15 mp-6ea4bff5 daemon.err smbd[519]: [1970/01/01 00:00:15.798456,  0] ../source3/lib/util_sock.c:455(open_socket_in)
Jan  1 00:00:15 mp-6ea4bff5 daemon.err smbd[519]:   open_socket_in(): socket() call failed: Address family not supported by protocol
Jan  1 00:00:15 mp-6ea4bff5 daemon.err smbd[519]: [1970/01/01 00:00:15.799247,  0] ../source3/smbd/server.c:690(smbd_open_one_socket)
Jan  1 00:00:15 mp-6ea4bff5 daemon.err smbd[519]:   smbd_open_once_socket: open_socket_in: Address family not supported by protocol
Jan  1 00:00:22 mp-6ea4bff5 user.notice motion: [0] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
Jan  1 00:00:22 mp-6ea4bff5 user.notice motion: [0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
Jan  1 00:00:22 mp-6ea4bff5 user.notice motion: [0] [NTC] [ALL] motion_startup: Motion trunkREV561 Started
Jan  1 00:00:22 mp-6ea4bff5 user.notice motion: [0] [NTC] [ALL] motion_startup: Logging to syslog
Jan  1 00:00:22 mp-6ea4bff5 user.notice motion: [0] [NTC] [ALL] motion_startup: Using default log type (ALL)
Jan  1 00:00:22 mp-6ea4bff5 user.notice motion: [0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (EMG)
Jan  1 00:00:22 mp-6ea4bff5 daemon.err dhclient: receive_packet failed on eth0: Network is down
Jan  1 00:00:25 mp-6ea4bff5 user.notice ethernet: disconnected
Jan  1 00:00:25 mp-6ea4bff5 user.notice network: eth0 has no IP address
Jan  1 00:00:28 mp-6ea4bff5 daemon.info init: starting pid 655, tty '': '/sbin/userinit'
Jan  1 00:00:28 mp-6ea4bff5 daemon.info init: starting pid 656, tty '/dev/tty1': '/sbin/getty -L tty1 115200 vt100'
Jan  1 00:00:30 mp-6ea4bff5 user.notice ethernet: disconnected
Jan  1 00:00:30 mp-6ea4bff5 user.notice network: eth0 has no IP address
Jan  1 00:00:35 mp-6ea4bff5 user.notice ethernet: disconnected
Jan  1 00:00:35 mp-6ea4bff5 user.notice network: eth0 has no IP address
Jan  1 00:00:41 mp-6ea4bff5 user.notice ethernet: disconnected
Jan  1 00:00:41 mp-6ea4bff5 user.notice network: eth0 has no IP address
Jan  1 00:00:46 mp-6ea4bff5 user.notice ethernet: disconnected for 20 seconds, rebooting
Jan  1 00:00:46 mp-6ea4bff5 user.notice network: eth0 had no IP address for 20 seconds, rebooting
Jan  1 00:00:46 mp-6ea4bff5 daemon.info init: starting pid 821, tty '/dev/null': '/etc/init.d/rcK'
Jan  1 00:00:47 mp-6ea4bff5 syslog.info syslogd exiting
Jan  1 00:00:02 mp-6ea4bff5 syslog.info syslogd started: BusyBox v1.23.2
Jan  1 00:00:04 mp-6ea4bff5 auth.err passwd: /etc/passwd: Read-only file system
Jan  1 00:00:04 mp-6ea4bff5 auth.err passwd: can't update password file /etc/passwd
Jan  1 00:00:04 mp-6ea4bff5 auth.err passwd: /etc/passwd: Read-only file system
Jan  1 00:00:04 mp-6ea4bff5 auth.err passwd: can't update password file /etc/passwd
Jan  1 00:00:11 mp-6ea4bff5 auth.info sshd[465]: Server listening on 0.0.0.0 port 22.
Jan  1 00:00:13 mp-6ea4bff5 daemon.notice proftpd[484]: localhost - ProFTPD 1.3.4d (maint) (built Sun Jul 19 2015 19:35:27 EEST) standalone mode STARTUP
Jan  1 00:00:13 mp-6ea4bff5 cron.info crond[492]: crond (busybox 1.23.2) started, log level 8
Jan  1 00:00:13 mp-6ea4bff5 daemon.info dhclient: DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 6
Jan  1 00:00:13 mp-6ea4bff5 daemon.info dhclient: DHCPREQUEST on eth0 to 255.255.255.255 port 67
Jan  1 00:00:13 mp-6ea4bff5 daemon.info dhclient: DHCPOFFER from 192.168.1.254
Jan  1 00:00:13 mp-6ea4bff5 daemon.info dhclient: DHCPACK from 192.168.1.254
Jan  1 00:00:14 mp-6ea4bff5 daemon.info dhclient: bound to 192.168.1.173 -- renewal in 41131 seconds.
Jan  1 00:00:15 mp-6ea4bff5 daemon.err smbd[519]: [1970/01/01 00:00:15.556912,  0] ../lib/util/become_daemon.c:124(daemon_ready)
Jan  1 00:00:15 mp-6ea4bff5 daemon.err smbd[519]:   STATUS=daemon 'smbd' finished starting up and ready to serve connections
Jan  1 00:00:15 mp-6ea4bff5 daemon.err smbd[526]: [1970/01/01 00:00:15.618825,  0] ../source3/printing/print_standard.c:69(std_pcap_cache_reload)
Jan  1 00:00:15 mp-6ea4bff5 daemon.err smbd[526]:   Unable to open printcap file /etc/printcap for read!
Jan  1 00:00:15 mp-6ea4bff5 daemon.err nmbd[523]: [1970/01/01 00:00:15.626812,  0] ../lib/util/become_daemon.c:124(daemon_ready)
Jan  1 00:00:15 mp-6ea4bff5 daemon.err nmbd[523]:   STATUS=daemon 'nmbd' finished starting up and ready to serve connections
Jan  1 00:00:15 mp-6ea4bff5 daemon.err smbd[519]: [1970/01/01 00:00:15.713583,  0] ../source3/lib/util_sock.c:455(open_socket_in)
Jan  1 00:00:15 mp-6ea4bff5 daemon.err smbd[519]:   open_socket_in(): socket() call failed: Address family not supported by protocol
Jan  1 00:00:15 mp-6ea4bff5 daemon.err smbd[519]: [1970/01/01 00:00:15.714613,  0] ../source3/smbd/server.c:690(smbd_open_one_socket)
[... cut ...]
    [root@mp-6ea4bff5 log]# 

motion.log 
    [root@mp-6ea4bff5 log]# cat motion.log 
    [0] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
    [0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
    [0] [NTC] [ALL] motion_startup: Motion trunkREV561 Started
    [0] [NTC] [ALL] motion_startup: Logging to syslog
    [0] [NTC] [ALL] motion_startup: Using default log type (ALL)
    [0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (EMG)
    [root@mp-6ea4bff5 log]#

redfox05, 2015-08-02:
FYI, I have the original full log files if you want me to email them. I backed up the entire log directory.

redfox05, 2015-08-02:
I wonder if its down to having a first generation model B pi....

ccrisan, 2015-08-02:
dmesg.log, please

redfox05, 2015-08-02:
Cant find dmesg.log.....
[image]

ccrisan, 2015-08-02:
My apologies, it's actually the output of the command "dmesg". You can create it using " dmesg > /data/log/dmesg.log. Unfortunately you need to run this command right after a problematic boot, on your PI.

Anyways, it looks like your wifi connection fails to start, and it has nothing to do with your static IP. You need to check your network name and key in /data/etc/wpa_supplicant.conf and to make sure you use a wpa-encrypted network.

redfox05, 2015-08-02:
[root@mp-6ea4bff5 ~]# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 3.18.12-quick (ccrisan@supermicro) (gcc version 4.8.4 (Buildroot 2015.08-git-g73e1681) ) #1 PREEMPT Sun Jul 19 19:51:43 EEST 2015
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] Machine: BCM2708
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 32768
[    0.000000] free_area_init_node: node 0, pgdat c0515b48, node_mem_map c7ef8000
[    0.000000]   Normal zone: 256 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708.boardrev=0x2 bcm2708.serial=0x6ea4bff5 smsc95xx.macaddr=B8:27:EB:A4:BF:F5 bcm2708_fb.fbswap=1 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0xec00000 vc_mem.mem_size=0x10000000  dwc_otg.fiq_fix_enable=1 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait ro rootflags=noload
[    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Memory: 124356K/131072K available (4159K kernel code, 201K rwdata, 688K rodata, 128K init, 328K bss, 6716K reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
                   vmalloc : 0xc8800000 - 0xff000000   ( 872 MB)
                   lowmem  : 0xc0000000 - 0xc8000000   ( 128 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0xc0008000 - 0xc04c3f0c   (4848 kB)
                     .init : 0xc04c4000 - 0xc04e4000   ( 128 kB)
                     .data : 0xc04e4000 - 0xc0516580   ( 202 kB)
                      .bss : 0xc0516580 - 0xc0568824   ( 329 kB)
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:522
[    0.000022] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483648000ns
[    0.000067] Switching to timer-based delay loop, resolution 1000ns
[    0.000627] Console: colour dummy device 80x30
[    0.001488] console [tty1] enabled
[    0.001535] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=1000)
[    0.001598] pid_max: default: 32768 minimum: 301
[    0.001912] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.001967] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.002894] CPU: Testing write buffer coherency: ok
[    0.003352] Setting up static identity map for 0x415de8 - 0x415e20
[    0.005385] devtmpfs: initialized
[    0.022133] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.023118] NET: Registered protocol family 16
[    0.028274] DMA: preallocated 4096 KiB pool for atomic coherent allocations
[    0.029281] bcm2708.uart_clock = 3000000
[    0.030988] mailbox: Broadcom VideoCore Mailbox driver
[    0.031134] bcm2708_vcio: mailbox at f200b880
[    0.031509] bcm_power: Broadcom power driver
[    0.031559] bcm_power_open() -> 0
[    0.031589] bcm_power_request(0, 8)
[    0.532281] bcm_mailbox_read -> 00000080, 0
[    0.532324] bcm_power_request -> 0
[    0.532525] Serial: AMBA PL011 UART driver
[    0.532711] dev:f1: ttyAMA0 at MMIO 0x20201000 (irq = 83, base_baud = 0) is a PL011 rev3
[    0.568663] usbcore: registered new interface driver usbfs
[    0.568845] usbcore: registered new interface driver hub
[    0.568973] usbcore: registered new device driver usb
[    0.569086] media: Linux media interface: v0.10
[    0.569274] Linux video capture interface: v2.00
[    0.570969] cfg80211: Calling CRDA to update world regulatory domain
[    0.571706] Switched to clocksource stc
[    0.572142] FS-Cache: Loaded
[    0.572506] CacheFiles: Loaded
[    0.586302] NET: Registered protocol family 2
[    0.587568] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[    0.587659] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
[    0.587780] TCP: Hash tables configured (established 1024 bind 1024)
[    0.587898] TCP: reno registered
[... 81 lines cut ...]
[    2.289932] usb 1-1: new high-speed USB device number 2 using dwc_otg
[    2.292511] Indeed it is in host mode hprt0 = 00001101
[    2.412239] EXT4-fs (mmcblk0p3): recovery complete
[    2.416530] EXT4-fs (mmcblk0p3): mounted filesystem with ordered data mode. Opts: (null)
[    2.483169] usb 1-1: New USB device found, idVendor=0424, idProduct=9512
[    2.486764] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    2.493316] hub 1-1:1.0: USB hub found
[    2.500820] hub 1-1:1.0: 3 ports detected
[    2.642984] : bcm2835-v4l2: error -1 while loading driver
[    2.779955] usb 1-1.1: new high-speed USB device number 3 using dwc_otg
[    2.877240] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00
[... cut ...]
[    3.546369] bcm2708_i2c bcm2708_i2c.0: BSC0 Controller at 0x20205000 (irq 79) (baudrate 100000)
[    3.605156] smsc95xx v1.0.4
[    3.678821] random: nonblocking pool is initialized
[    3.683997] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-bcm2708_usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:a4:bf:f5
[    3.717320] usbcore: registered new interface driver smsc95xx
[    5.310642] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    6.850793] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[root@mp-6ea4bff5 ~]#

redfox05, 2015-08-02:
Not entirely sure if it classes as a problematic boot, as they tend to crash before I get a chance to login :P

ccrisan, 2015-08-02:
Your wlan adapter resets every now and then. It may be a power problem. If you're using an older rpi model, there are known power issues when it comes to usb devices. Try using a newer board or a power adapter that supplies 5.25V.

redfox05, 2015-08-02:
But thats strange. I am connecting with a 1.5A adapter AND there is no WiFi adapter in there currently. I unplugged it when I put the new image in.

redfox05, 2015-08-02:
I appreciate your help btw. Looks like I will indeed just have to upgrade. The problem is that I've got a great water/weatherproof housing for this one in the form of the PiCE. They dont make them for the newer models. Any suggestions for outdoor housings? :(

ccrisan, 2015-08-02:
You need to have your wifi adapter connected if you have configured it in the UI. If you don't need the wifi adapter anymore, disable the wireless network from the UI first. Otherwise you'll be stuck in a reboot loop.

redfox05, 2015-08-02:
Yup, understood. Although I am now using a brand new image downloaded from your site, and flashed that to the pi after re-flashing the SD. The default image from your site is set-up for ethernet is it not?

ccrisan, 2015-08-03:
Yes, the image will automatically detect the presence of a connected ethernet cable and will use DHCP. WiFi is disabled by default, but, when enabled and configured, it must work; otherwise the system will continuously reboot until managing to connect to the wireless network.

pandrew, 2015-08-05:
I have a couple of pis and i know by experience that if you cant supply enough juce it will reboot. Some can handle a wifi adapter and not crash but sticking in a microsoft keyboard it either cant power the keyboard or it reboots. I have the same issue with a pi1 here running noir and motionpie. If i connected two clients to the web it reboots, i have switched to the official powersupply and theres no issues... yet.

redfox05, 2015-08-05:
Interesting. Which is the "official" power supply? I presume if we match
the Amps and Voltage that should be fine.

pandrew, 2015-08-05:
@redfox05 https://www.raspberrypi.org/products/universal-power-supply/

ptorstensson, 2015-08-15:
I must reply to this, first thanks for a fabulous dist.. 

Had problems scratching my head off with the latest 20150719-release, also tried the 20150621-release. But SEMC95xx, Network, found USB-device.. 

My NoIR-cam has two IR-Led's at 2W totally. I fed my RPi with 5.3V, 4.2Amps. Eternal bootloop :[ 

Disconnected one of the IR-Leds, so now only 1W usage plus the standard RPi. Works like a charm now. So for me, my powersource could not deliver what the quite sensitive RPi required. 
My basic knowledge of electrical maths summed up that it should work with 2*1W LED's, but no. 

Good luck..

scohrab, 2015-08-16:
Just had this problem - rebooting when trying to use WiFi. bootlog file showed constant reboots. I turned off the camera LED (in 'Expert Settings', set 'Enable CSI Camera LED' to Off) and that fixed it. I now have working WiFi connection. Perhaps that LED just tips the power consumption over the edge for some systems ?

clairmont, 2015-08-21:
I had this issue, or something similar. I would check my router's connected devices and see the raspberry pi camera's name would appear for a few seconds and then disappear. I am using a Raspberry Pi 2 B / 5MP camera with IR / 2a power cable - I solved the issue after a few flashes of the image and have had a stable 24+ hour uptime so far. Try re-imaging until you get a good / stable one! -- Use Win32 Disk Imager and the SD Card utility to format the card!

redfox05, 2015-08-21:
That's weird, I thought an image is an image..... How can they differ