ccrisan/motionpie #261: Connection to camera lost (and green screen)

shoxcorp, 2015-08-08:
Hey guys,
I have a D-Link DCS-930LB1 network camera which after connecting to motionPie works. However after i change the name of the camera and alter any of the camera's settiongs (motion detection, streaming etc) after a couple of seconds the screen turns green and afterwards grey, with the message that the connection to the camera was lost.

If i try to access the web interface of the camera itself it all works normally without problems. also the cgi page of the camera works on the browser.

Any ideas?

Thanks!
Harry

ccrisan, 2015-08-09:
Does your camera supply JPEG snapshots or a MJPG stream?

shoxcorp, 2015-08-09:
JPEG snapshots it is

ccrisan, 2015-08-09:
Could you send me the motion.log and motioneye.log files (downloadable from the UI)?

VeggieVampire, 2016-05-23:
Having the same issue.

ccrisan, 2016-05-23:
What version?