ccrisan/motionpie #262: "Error occurred... refresh is recommended"

cy80rg101, 2015-08-09:
Just got MotionPie installed on RPi2 connecting to two local ip cams. 

Had both cameras configged and connected fine. 

Enabled "Ping host to monitor connection" bit (pings what looks like one of Googles DNS' to confirm connection, attempt reconnect if not). 

Rebooted. 

Now cant get back into motionPie - comes up with login - I login as admin.

"Error occured... refresh is recommended" comes up - followed by "loading chevrons". 

Looks like it times out, I have the blacl mP background, logo top right, key icon top right, and the copyright info at the bottom. 

However, the content in the middle never loads, I have a "broken image" icon top left. 

Will format and reinstall the image and re-config from scratch.

ccrisan, 2015-08-09:
It would be great if you could recover the logs from such a broken setup. Would you by any chance have a Linux PC with an SD card reader? If so, please send me an archive of the log directory from the data partition.

cy80rg101, 2015-08-09:
Darn it...

Rebuilt the sd - got it all up and running, tweaked the config, all running well. 

Finally thought I'd add the icing on the cake by re-enabling the connection ping test - enabled the function using the default 8.8.8.8 , rebooted, and Lo! Does the same dang thing... grr! ;)

Applied the setting, logged in - sat waiting on the "loading chevrons". 

Will give it a good 30 mins to see if it ever comes back. 

Unfortunately, I dont have a full linux box - PC user with an Rpi. 

Anything else I can do to provide more info however, please just let me know and I'll see if I can do. 

(Great image btw! :) thanks for all your hard work on bringing this out...)

cy80rg101, 2015-08-09:
FWIW - I did enable the Expert settings > Overclocking >RPI2 (which I assume is correct to overclock / make use of the RPI2 h/w?).

First time saw this issue tho - I hadnt done this. 

RPI2 running wired cat5.
Two IP Cams (1xtenvis, 1xWansview) on wifi
65gb sd on Pi2

cy80rg101, 2015-08-09:
Re the logs - I should be able to SSH in from my PC, I assume I could get the logs that way?

ccrisan, 2015-08-09:
Well you could recover the logs using SSH/SFTP by copying the /data/log/ directory over to your (windows-based?) PC. WinSCP or PSCP should do the job.

cy80rg101, 2015-08-09:
k - will try that...

FWIW - i think I briefly saw the "Error occurred...refesh" msg very quickly, then the chevrons. 

Refreshing either times out, or comes back with the login / missing content / chevrons again. 

Will see if I can get those logs - want me to upload them here somewhere?

cy80rg101, 2015-08-09:
Planning to SSH in using Bitvise SSH Client, then using the mP Admin user to login.

This should work?

What's the SSH port for mP - 22?

Hmm, getting "actively refused it" responce from mP... ?

cy80rg101, 2015-08-09:
K - can connect to mP, I get a "User Authentication Banner" with a welcome message from mP instance.

How do I get the file from here - doesnt seem to be any ui ?

ccrisan, 2015-08-09:
That's why I recommended WinSCP :) Try WinSCP and you'll find it alright when it comes to copying files from and to your SSH-enabled devices.

ccrisan, 2015-08-09:
By the way, I'm currently working on improving the network camera support in motionEye (the GUI used by motionPie) and I would really like to get all the netcam-related problems fixed in this next release. Are you by any chance available on Google Hangouts, or Skype or whatever so we can debug your issue easier?

cy80rg101, 2015-08-09:
Yes - i'm more than up for helping out where I can!

Pretty new with *nix systems, but work in online and have been hacking things around for years ;)

Have rebuilt the image now (and got all 3 cams up and running) - anything you need me to do, just let me know.

ccrisan, 2015-08-09:
Looking over your issue report again, here are some ideas:
* the default 8.8.8.8:80 address/port don't really work - leaving it to that will only generate reboot loops (the new version will have `www.google.com` as default host)
* overclocking the PI, especially the PI2 could cause issues, and I'm not talking about h/w damages, but rather illegal instructions and other CPU/memory faults

I'll close the issue now, but should the problem reappear, feel free to reopen it.

closed 2015-08-09 by ccrisan