ccrisan/motionpie #263: Include uvcdynctrl to handle unruly webcams

TravisCHoward, 2015-08-12:
There are at least two models of Microsoft Lifecam webcams that continually refocus for seemingly no reason. uvcdynctrl allows the autofocus to be disabled. Even if you don't include configuration in the web GUI, please include the application so that owners of these crappy webcams can make them usable with motionpie. There are more details here: http://www.bot-thoughts.com/2013/01/lifecam-hd-6000-autofocus-fix-raspberry.html

ccrisan, 2015-08-12:
Including uvcdynctrl is something that makes sense. I'll see if I can compile it for ARM. Nevertheless, executing scripts at startup is achievable using /data/etc/userinit.sh. See [Adding Custom Startup Commands](https://github.com/ccrisan/motionpie/wiki/Tweaks#adding-custom-startup-commands).

ccrisan, 2015-08-16:
libwebcam/uvcdynctrl support has been added

closed 2015-08-16 by ccrisan