ccrisan/motionpie #264: TP-Link TL-WN725N V2 not working, leading to reboot

Narno38, 2015-08-13:
Hello,
First thing, thanks a lot for the hard work and excellent development!
I am trying to use the TP-LINK USB WiFi Dongle with motionPie (this one is based on realtek 8188eu and works fine under Raspbian provided a few commands). When using with motionPie I saw the message that module is from the staging directory, I was wondering which other USB Wifi dongle could be used then?
Basically if I try to plug in the dongle before booting, the system will permanently reboot until I remove the dongle. If trying to insert it after boot, it is found correctly but then how can I get the wlan0 interface started? Would this be a buildroot issue or one for motionPie? Any idea?
Thanks again,
Arnaud

ccrisan, 2015-08-13:
It probably is a motionPie problem. If support for your dongle exists in Raspbian, just hold on for a few more weeks until I manage to release the new version of motionPie, based on the latest Raspbian kernel.

Narno38, 2015-08-13:
Excellent, thanks for the fast response, looking forward to any update, happy to provide more inputs as required!

kan-uddf, 2015-08-25:
just a comment... narno38 it is workable in raspbian if you download and install your own module as teh built in one still wont work so it shouldn't be too hard to fix up from scratch here, but for some reason i cant get it working might be kernel version differences not sure, hopefully ccrisan gets it working soon until than ill run wired I guess..

BaGRoS, 2015-10-06:
I have tested, it works for me, exactly the same model.

kan-uddf, 2015-10-06:
hrmm an update hasn't come out since before august, wonder why yours works.

ill test again see what else i can see in a few days.

adaniel81, 2015-10-07:
hello to everyone, I have the same problem with the wifi module TP-Link TL-WN725N V2, the system restarts continuously until you disconnect the device, on raspbian working properly, I have the version of 20150719 motionpie
Thanks for the great work

Narno38, 2015-11-28:
Just some news here. The latest version works fine and allowed me to test various configurations. It turned out that (for some reason) there are several versions of the TP-Link TL-WN725N Nano :

*         The silver one : http://www.amazon.fr/ADAPTER-150MBPS-WIRELESS-TL-WN725N-TP-LINK/dp/B0118KLWKM/ref=sr_1_3?ie=UTF8 <http://www.amazon.fr/ADAPTER-150MBPS-WIRELESS-TL-WN725N-TP-LINK/dp/B0118KLWKM/ref=sr_1_3?ie=UTF8&qid=1448720603&sr=8-3&keywords=WN725N> &qid=1448720603&sr=8-3&keywords=WN725N

*         The golden one : http://www.amazon.fr/TP-Link-TL-WN725N-Nano-adaptateur-sans/dp/B008IFXQFU/ref=sr_1_1?ie=UTF8 <http://www.amazon.fr/TP-Link-TL-WN725N-Nano-adaptateur-sans/dp/B008IFXQFU/ref=sr_1_1?ie=UTF8&qid=1448720603&sr=8-1&keywords=WN725N> &qid=1448720603&sr=8-1&keywords=WN725N 

 

While both show the same form factor and engravings, the golden one (released at a later date) seems by far more stable. With this one, motioneyeos works like a charm, while it still reboots with the silver one… a behavior I could reproduce with the latest Raspian Jessie as well.

 

Thanks !

 

L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

ccrisan, 2015-11-28:
Thanks for keeping us posted. Wifi adapters are always a mess; that's why I strongly suggest using a wired connection, where possible. This is, at least, how I use motionEyeOS with my 6 cameras, and I never had a connection problem with them.

closed 2015-11-28 by ccrisan