ccrisan/motionpie #265: Frozen image - An error occurred. Refreshing is recommended.

SuperSharpShooter, 2015-08-14:
This morning I was getting lots of notifications from the camera being triggered by rain on the window, so I turned email notifications off. Since then the camera has been frozen.

Any time I change any items in the UI, I get the "An error occurred. Refreshing is recommended." message. Can't restore from a fresh config backup either. Rebooting doesn't help.

I'm running a week old install of 20150719

In Motioneye.log there's a lot of this:

2015-08-14 11:21:27: WARNING: Connect error on fd 12: ECONNREFUSED
2015-08-14 11:21:27: ERROR: connection problem detected for mjpg client for camera 1 on port 8081
2015-08-14 11:21:28: ERROR: [Errno 12] Cannot allocate memory
Traceback (most recent call last):
  File "home/ccrisan/motionpie/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1302, in _stack_context_handle_exception
  File "home/ccrisan/motionpie/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/stack_context.py", line 314, in wrapped
  File "/programs/motioneye/src/motionctl.py", line 124, in start
    process = subprocess.Popen(args, stdout=log_file, stderr=log_file, close_fds=True, cwd=settings.CONF_PATH)
  File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
  File "/usr/lib/python2.7/subprocess.py", line 1231, in _execute_child
OSError: [Errno 12] Cannot allocate memory

SuperSharpShooter, 2015-08-14:
Did a hard reboot and it looks like it's running again...

ccrisan, 2015-08-14:
Yeah, it looks like an out-of-memory problem. There's not much I can do about it. Just reduce the various stream parameters (resolution, refresh rate) or use a RPI2.

closed 2015-08-14 by ccrisan