ccrisan/motionpie #266: Email notifications

reg44reg, 2015-08-19:
I just setup my Pi and the only piece I haven't been able to get working is the email notifications on motion detection. The email account I set has been tested and works and I've double check the smtp servers with another service and they're correct.

Thanks for any help
Reg

My log files:
http://s000.tinyupload.com/?file_id=27950266415946541759

ccrisan, 2015-08-19:
From your motion.log:

2015-08-19 11:57:31: ERROR: failed to send mail: timed out
Traceback (most recent call last):
  File "/programs/motioneye/sendmail.py", line 182, in on_message
    send_mail(server, port, account, password, tls, to, subject, message,
files)
  File "/programs/motioneye/sendmail.py", line 57, in send_mail
    conn = smtplib.SMTP(server, port, timeout=getattr(settings,
'SMTP_TIMEOUT', 60))
  File "/usr/lib/python2.7/smtplib.py", line 256, in __init__
  File "/usr/lib/python2.7/smtplib.py", line 316, in connect
  File "/usr/lib/python2.7/smtplib.py", line 291, in _get_socket
  File "/usr/lib/python2.7/socket.py", line 571, in create_connection
timeout: timed out

The SMTP connection failed after 60 seconds of waiting. Make sure that your
PI has full internet connection and that it can actually reach your SMTP
server on port 465 or 587 or whatever port you're using.

Calin.

la5cq, 2016-05-18:
From motioneye log:
2016-05-18 08:57:06: [sendmail]    ERROR: failed to send mail: Connection unexpectedly closed: timed out
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 208, in on_message
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 52, in send_mail
  File "/usr/lib/python2.7/smtplib.py", line 256, in __init__
  File "/usr/lib/python2.7/smtplib.py", line 317, in connect
  File "/usr/lib/python2.7/smtplib.py", line 365, in getreply
SMTPServerDisconnected: Connection unexpectedly closed: timed out 

My configuration in motioneye is correct but there might be some problem around the TLS/SSL option.
I wrote a python test program to verify my parameters and I have to call smtplib.SMTP_SSL('xxx.xxx.no',465) and not smtplib.SMTP('xxx.xxx.no',465) in order to connect to my email server.

What can I do to achieve the same in motioneyes?

Per.