ccrisan/motionpie #269: Adding second camera

drknstrmyknight, 2015-08-23:
I get a 
ERROR: failed to list remote cameras on http://192.168.111.60:8081: request timed out

 when adding a second camera as a motionpi cam, though I can add it as a network camera just fine.

The second cam only shows this:
2015-08-23 07:44:56: INFO: hello! this is motionPie 20150719
2015-08-23 07:44:59: INFO: cleanup started
2015-08-23 07:44:59: INFO: wsswitch started
2015-08-23 07:44:59: INFO: thumbnailer started
2015-08-23 07:45:00: INFO: server started
2015-08-23 07:45:13: WARNING: control hue not found for device /dev/video0
2015-08-23 07:45:13: WARNING: control hue not found for device /dev/video0
2015-08-23 07:45:13: WARNING: control hue not found for device /dev/video0

They are on the same subnet and I can get to and control them individually

ccrisan, 2015-08-23:
I think you tried to add a remote motionEye camera using a MJPEG url. Use the normal motionEye url (http://192.168.111.60)

closed 2015-08-23 by ccrisan

drknstrmyknight, 2015-08-24:
I threw a new image on both pi's. (1B+ with raspperry cams).When I attempt to add a second Motion Pi instance, I get the error occurred, refreshThe local cam is gone and cannot be added.

   
Closed #269.—

ccrisan, 2015-08-24:
Please send me the [logs](https://github.com/ccrisan/motionpie/wiki/Reporting-An-Issue#attach-log-files).