ccrisan/motionpie #270: Double Text Overlay when Video Rotation changed

rockman55, 2015-08-23:
Is there some way to not have double Text Overlay on the front end when the Video Rotation is changed? Please see screen shot for WebCam.
Thanks,
Danny
[image]

ccrisan, 2015-08-23:
motionEye sets text_double to True when width > 320. It's about the resolution and not the orientation. Of course there's no perfect compromise and I wish there were more options regarding the text size, but this is what we have for now.

rockman55, 2015-08-26:
I went into the mmal conf file and changed the rotation there. Thanks for your help.

Danny