ccrisan/motionpie #276: Not recording when network is down.

cegla, 2015-08-28:
I found motionpie as very interesting project. 
I'd like to use it as a standalone recording device. Unfurtunately when I try to record when my PI2 device is disconnected from network (wheter Wifi or Ethernet, both interfaces are down) nothing happens (now new recordings). As I only connect the Ethernet interface recording starts even though IP address is not assigned (just eth0 interface is up).
What may be the problem? Is it some configuration issue? I store recordings on PI2's local partition (no network communication needed).
Thanks in advance
Piotr

ccrisan, 2015-08-28:
motionPie doesn't work properly without a network connection. If you really need it to work without any kind of connectivity, make sure you *don't* have wifi enabled in UI settings and boot it *without* wifi adapter or ethernet cable plugged in. It should work just fine. If it doesn't, connect it to a monitor and look for suspicious error messages.

cegla, 2015-08-28:
Thanks for quick response!
Without ethernet cable and without wifi adapter works perfectly. (With Wifi adapter and no ethernet cable plugged - doesn't, event when booted so).
So for me the solution was to remove Wifi.
Thank you very much! Great project.