ccrisan/motionpie #277: Access to MotionPie via Reverse Proxy with Nginx

nitschkecm, 2015-08-28:
I have tried to access my Motionpie via a reverse proxy I have. It works fine when I am using the following config 

location /cams {
                proxy_pass http://192.XXX.XXX.XXX:8081;
                proxy_set_header Host $host;
                proxy_set_header X-Real-IP $remote_addr;
                proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        }

However that lets me just see the stream of one of the cameras. When I try to point the proxy pass to 
proxy_pass http://192.XXX.XXX.XXX/;

It does not work. I am sure there there must be a simple trick which I am not able to see. Any help is more than welcome.

ccrisan, 2015-08-30:
How about `proxy_pass http://192.x.y.z;` (note there's no trailing slash).

nitschkecm, 2015-08-30:
Hi for some very strange reason it does not work, 

when I do exactly that I get

{"error": "not found"}

If you have any suggestion I would much appreciate. 

Thanks !

Carsten

nitschkecm, 2015-09-15:
Has anybody gotten this to work by chance ? I have tried all different ways I could think of with / and without, with Port 80 or without, but here has not been a way to get this to work. Any suggestions are more than welcome !

ccrisan, 2015-09-15:
It doesn't work and needs to be fixed upstram (in motionEye).

ccrisan, 2015-09-15:
It should now be fixed upstream. The fix will be included with the next release.

closed 2015-09-15 by ccrisan