ccrisan/motionpie #278: Adding a 3rdcam

drknstrmyknight, 2015-08-29:
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 3.18.12-quick (ccrisan@supermicro) (gcc version 4.8.4 (Buildroot 2015.08-git-g73e1681) ) #1 PREEMPT Sun Jul 19 19:51:43 EEST 2015
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] Machine: BCM2708
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 98304
[    0.000000] free_area_init_node: node 0, pgdat c0515b48, node_mem_map d7cf8000
[    0.000000]   Normal zone: 768 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 98304 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 97536
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708.boardrev=0xe bcm2708.serial=0xb41cc420 smsc95xx.macaddr=B8:27:EB:1C:C4:20 bcm2708_fb.fbswap=1 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.fiq_fix_enable=1 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait ro rootflags=noload
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 384156K/393216K available (4159K kernel code, 201K rwdata, 688K rodata, 128K init, 328K bss, 9060K reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
                   vmalloc : 0xd8800000 - 0xff000000   ( 616 MB)
                   lowmem  : 0xc0000000 - 0xd8000000   ( 384 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0xc0008000 - 0xc04c3f0c   (4848 kB)
                     .init : 0xc04c4000 - 0xc04e4000   ( 128 kB)
                     .data : 0xc04e4000 - 0xc0516580   ( 202 kB)
                      .bss : 0xc0516580 - 0xc0568824   ( 329 kB)
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:522
[    0.000021] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483648000ns
[    0.000068] Switching to timer-based delay loop, resolution 1000ns
[    0.000634] Console: colour dummy device 80x30
[    0.001491] console [tty1] enabled
[    0.001536] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=1000)
[    0.001600] pid_max: default: 32768 minimum: 301
[    0.001927] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.001983] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.002915] CPU: Testing write buffer coherency: ok
[    0.003369] Setting up static identity map for 0x415de8 - 0x415e20
[    0.005397] devtmpfs: initialized
[    0.022172] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.023154] NET: Registered protocol family 16
[    0.028310] DMA: preallocated 4096 KiB pool for atomic coherent allocations
[    0.029310] bcm2708.uart_clock = 3000000
[    0.031012] mailbox: Broadcom VideoCore Mailbox driver
[    0.031158] bcm2708_vcio: mailbox at f200b880
[    0.031533] bcm_power: Broadcom power driver
[    0.031583] bcm_power_open() -> 0
[    0.031613] bcm_power_request(0, 8)
[    0.532305] bcm_mailbox_read -> 00000080, 0
[    0.532348] bcm_power_request -> 0
[    0.532548] Serial: AMBA PL011 UART driver
[    0.532735] dev:f1: ttyAMA0 at MMIO 0x20201000 (irq = 83, base_baud = 0) is a PL011 rev3
[    0.568689] usbcore: registered new interface driver usbfs
[    0.568868] usbcore: registered new interface driver hub
[    0.568997] usbcore: registered new device driver usb
[    0.569111] media: Linux media interface: v0.10
[    0.569299] Linux video capture interface: v2.00
[    0.570983] cfg80211: Calling CRDA to update world regulatory domain
[    0.571724] Switched to clocksource stc
[    0.572157] FS-Cache: Loaded
[    0.572525] CacheFiles: Loaded
[    0.586337] NET: Registered protocol family 2
[    0.587595] TCP established hash table entries: 4096 (order: 2, 16384 bytes)
[    0.587794] TCP bind hash table entries: 4096 (order: 2, 16384 bytes)
[    0.587927] TCP: Hash tables configured (established 4096 bind 4096)
[    0.588038] TCP: reno registered
[    0.588084] UDP hash table entries: 256 (order: 0, 4096 bytes)
[... 83 lines cut ...]
[    2.565762] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    2.571931] hub 1-1:1.0: USB hub found
[    2.575822] hub 1-1:1.0: 5 ports detected
[    2.625834] bcm2835-v4l2: scene mode selected 0, was 0
[    2.626397] bcm2835-v4l2: V4L2 device registered as video0 - stills mode > 2592x1944
[    2.636069] bcm2835-v4l2: Broadcom 2835 MMAL video capture ver 0.0.2 loaded.
[    2.856053] usb 1-1.1: new high-speed USB device number 3 using dwc_otg
[    2.945247] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00
[    2.949400] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    3.085596] udevd[82]: starting version 3.0
[    3.107373] random: udevd urandom read with 110 bits of entropy available
[    3.131685] udevd[82]: specified group 'input' unknown
[    3.499985] bcm2708_i2c_init_pinmode(1,2)
[    3.514451] bcm2708_i2c_init_pinmode(1,3)
[    3.531900] bcm2708_i2c bcm2708_i2c.1: BSC1 Controller at 0x20804000 (irq 79) (baudrate 100000)
[    3.612096] smsc95xx v1.0.4
[    3.688421] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-bcm2708_usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:1c:c4:20
[    3.724346] usbcore: registered new interface driver smsc95xx
[    3.817110] random: nonblocking pool is initialized
[   32.121438] EXT4-fs (mmcblk0p3): mounted filesystem with ordered data mode. Opts: (null)
[   33.024690] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   34.572631] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x45E1
[  236.805817] bcm2835-v4l2: error 0 waiting for frame completion
[  259.884829] bcm2835-v4l2: error 0 waiting for frame completion
[336524.817838] bcm2835-v4l2: error 0 waiting for frame completion

drknstrmyknight, 2015-08-29:
Blanked the password from the 3rd cam, and when I added it it blew out the local camera and the remote that was working.
2015-08-24 13:41:52: INFO: hello! this is motionPie 20150719
2015-08-24 13:41:52: INFO: main config file /data/etc/motion.conf does not exist, using default values
2015-08-24 13:41:53: INFO: cleanup started
2015-08-24 13:41:53: INFO: wsswitch started
2015-08-24 13:41:53: INFO: thumbnailer started
2015-08-24 13:41:54: INFO: server started
2015-08-24 13:41:55: INFO: adding new camera with id 1...
2015-08-24 13:41:56: INFO: main config file /data/etc/motion.conf does not exist, using default values
2015-08-24 13:41:58: WARNING: control hue not found for device /dev/video0
2015-08-24 13:43:31: WARNING: 404 HEAD /robots.txt (192.168.111.100) 29.01ms
2015-08-24 13:44:03: WARNING: control hue not found for device /dev/video0
2015-08-24 13:44:03: WARNING: control hue not found for device /dev/video0
2015-08-24 13:44:03: WARNING: control hue not found for device /dev/video0
2015-08-24 13:44:48: WARNING: control hue not found for device /dev/video0
2015-08-24 13:44:48: WARNING: control hue not found for device /dev/video0
2015-08-24 13:45:12: WARNING: control hue not found for device /dev/video0
2015-08-24 13:45:12: WARNING: control hue not found for device /dev/video0
2015-08-24 13:59:36: WARNING: control hue not found for device /dev/video0
2015-08-24 13:59:36: WARNING: control hue not found for device /dev/video0
2015-08-24 13:59:36: WARNING: control hue not found for device /dev/video0
2015-08-24 16:17:30: WARNING: 404 HEAD /robots.txt (192.168.111.100) 28.27ms
2015-08-24 19:41:34: WARNING: 404 HEAD /robots.txt (192.168.111.100) 30.71ms
2015-08-24 19:48:53: WARNING: 404 HEAD /robots.txt (192.168.111.100) 58.40ms
2015-08-24 20:52:01: WARNING: 404 HEAD /robots.txt (192.168.111.100) 27.49ms
2015-08-26 22:09:30: WARNING: control hue not found for device /dev/video0
2015-08-26 22:10:22: WARNING: control hue not found for device /dev/video0
2015-08-26 22:10:27: WARNING: control hue not found for device /dev/video0
2015-08-26 22:10:28: WARNING: control hue not found for device /dev/video0
2015-08-26 22:10:28: WARNING: control hue not found for device /dev/video0
2015-08-26 22:10:28: WARNING: control hue not found for device /dev/video0
2015-08-27 06:20:32: WARNING: control hue not found for device /dev/video0
2015-08-27 10:24:38: WARNING: control hue not found for device /dev/video0
2015-08-27 10:24:39: WARNING: control hue not found for device /dev/video0
2015-08-27 10:24:39: WARNING: control hue not found for device /dev/video0
2015-08-27 10:26:36: WARNING: control hue not found for device /dev/video0
2015-08-27 10:26:39: WARNING: 404 HEAD /robots.txt (192.168.111.100) 27.55ms
2015-08-27 11:27:45: WARNING: 404 HEAD /robots.txt (192.168.111.100) 61.20ms
2015-08-27 21:25:33: WARNING: 404 HEAD /robots.txt (192.168.111.100) 27.66ms
2015-08-27 21:27:33: WARNING: 404 HEAD /robots.txt (192.168.111.100) 27.71ms
2015-08-27 21:47:47: WARNING: 404 HEAD /robots.txt (192.168.111.100) 60.09ms
2015-08-27 21:48:02: WARNING: 404 HEAD /robots.txt (192.168.111.100) 27.55ms
2015-08-27 21:48:14: WARNING: 404 HEAD /robots.txt (192.168.111.100) 27.58ms
2015-08-27 21:50:14: WARNING: 404 HEAD /robots.txt (192.168.111.100) 27.67ms
2015-08-27 21:50:27: WARNING: 404 HEAD /robots.txt (192.168.111.100) 43.03ms
2015-08-27 21:50:45: WARNING: 404 HEAD /robots.txt (192.168.111.100) 57.50ms
2015-08-27 21:52:28: WARNING: control hue not found for device /dev/video0
2015-08-27 22:10:00: WARNING: error on read
Traceback (most recent call last):
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/iostream.py", line 630, in _handle_read
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/iostream.py", line 600, in _read_to_buffer_loop
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/iostream.py", line 712, in _read_to_buffer
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/iostream.py", line 976, in read_from_fd
error: [Errno 113] No route to host
2015-08-27 22:10:00: ERROR: Uncaught exception
Traceback (most recent call last):
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/http1connection.py", line 691, in _server_request_loop
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/gen.py", line 807, in run
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/concurrent.py", line 209, in result
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/gen.py", line 810, in run
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/http1connection.py", line 166, in _read_message
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/gen.py", line 807, in run
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/concurrent.py", line 209, in result
  File "<string>", line 3, in raise_exc_info
error: [Errno 113] No route to host
2015-08-27 22:59:00: WARNING: 404 HEAD /robots.txt (192.168.111.100) 28.20ms
2015-08-27 22:59:39: WARNING: 404 HEAD /robots.txt (192.168.111.100) 37.39ms
[... cut ...]
    request = _make_request(scheme, host, port, username, password, uri + '/config/%(id)s/get/' % {'id': camera_id})
  File "/programs/motioneye/src/remote.py", line 50, in _make_request
    url += '&_signature=' + utils.compute_signature(method, url, data, password)
  File "/programs/motioneye/src/utils.py", line 396, in compute_signature
    key = _SIGNATURE_REGEX.sub('-', key)
TypeError: expected string or buffer
2015-08-29 20:39:10: ERROR: 500 GET /config/list/?_=1440880751078&_username=admin&_signature=cc91b4974f53bf37083a554e9350c3a78774ba30 (192.168.111.100) 76.29ms

ccrisan, 2015-08-30:
I need a clear explanation of what you did and what the problem was. Also, please send me the log files using online services like those indicated [here](https://github.com/ccrisan/motionpie/wiki/Reporting-An-Issue) rather than pasting everything in the issue content.

drknstrmyknight, 2015-08-30:
I just went to the ADD A CAMERA, selected remote MotionEye canera, put in HTTP://xxx.xxx.xxx.xxx, and poof, no cameras, plus I couldn't add the local camera back.

ccrisan, 2015-08-30:
Where did you select "remote MotionEye camera"? On what device? Do you run more than one motionPie-based devices? Do you have a PC running motionEye and trying to add your motionPie-based device to the PC? Please explain the entire scenario so I can understand what you were trying to do.

drknstrmyknight, 2015-08-30:
I have 3 Rasberry PI's running MotionPIOne one of these. when a try to add one of the other two as a remote MotionEye camera, it blows out the local camera as well as not adding the remote.Maybe using MotionEye as a place to view them all is the answer

ccrisan, 2015-08-30:
Well you should be able to add one motionPie to another one as motionEye, just as you seem to have tried. I will try to see if there's a problem with this scenario. Do all your motionPie devices run the same version, by the way?

Nevertheless, the correct solution is to configure a separate motionEye server used to centralize all of your devices.

drknstrmyknight, 2015-08-30:
They have have the latest version of MotionPI, all B+ with Raspberry Pi cams.Setting up MotionEye now, but ran into a snag as the motioneye.py file dated today is empty

ccrisan, 2015-08-30:
As I commented on the issue you just opened on motionEye, please stick to the bitbucket 0.26 version for now.

drknstrmyknight, 2015-10-14:
Did that and all is well, you can close this

closed 2015-10-14 by drknstrmyknight