ccrisan/motionpie #279: Web Hook support for authentication in URL

umbex, 2015-08-29:
Hi, 

everything works fine, but I've found that a web hook (GET) like this:
http://user:mypassword@myIP/mypath

return this error in motion.log:
ERROR: failed to call webhook: <urlopen error [Errno -3] Temporary failure in name resolution>

keeping the same url without authorization gives me the expected message:
ERROR: failed to call webhook: HTTP Error 401: Unauthorized

The same url, removing the required auth on target server, works fine. 

Best
umbex

ccrisan, 2015-08-30:
The webhook mechanism does not yet support authentication. I'll consider adding support for the syntax you described here.

umbex, 2015-08-30:
Excellent! Thank you, we enjoy your work.