ccrisan/motionpie #280: Support for USB Frame Grabber

maxholgasson, 2015-08-30:
Hello,

I have two different USB frame grabbers because I would like to use a analog video camera which has a bnc connector. One grabber has a utv007 chipset the other one a stk1160.
Is there any chance that I could use one of those with the beautiful motionpie?
http://www.amazon.de/dp/B002RX49AU  and the other grabber is http://www.amazon.de/dp/B0013BXFLG 

Thanks, Max

ccrisan, 2015-08-31:
If they have v4l support and they work with Raspbian, then yes, it's very likely they work with motionPie as well.

maxholgasson, 2015-08-31:
The Logilink works with the latest Raspbian on a Pi B+. I attached a screenshot of camorama
lsusb displays the device as following:
Bus 001 Device 064: ID 1b71:3002
It worked straight out of the box. I just installed rasbian with noobs and the camorama program

In MotionPie unfortunatly the device doesn't appear?  Do you have an idea what I could do?

Thanks, Max
[image]

ccrisan, 2015-09-01:
Hmm, it might be related to the fact that the latest Raspbian has switched to kernel 4.0 while motionPie is still with 3.18. Let's hope the driver will work with the next release (which will be based on kernel 4.0).

ccrisan, 2015-09-01:
Argh, nevermind my previous answer. It seems that the module has been left out from the kernel build, because it also requires sound support which is disabled in motionPie. [This](http://linuxtv.org/wiki/index.php/Easycap) is a nice page where all the required info has been put together.

I'll make sure to include this driver in the next release.

maxholgasson, 2015-09-02:
Hello, I got it working by manually installing motioneye. That's great! It broke a couple of times but at the moment it's running stable (i dont know anymore what I changed all).
Btw in which file could the values for the drop-down menu "Video resolution"  be changed or added some other?

ccrisan, 2015-09-02:
If your v4l device doesn't list any resolutions, motionEye uses utils.COMMON_RESOLUTIONS (see utils.py).

closed 2015-09-02 by ccrisan