ccrisan/motionpie #282: Movie playback issues

jpl90, 2015-08-31:
I've been using this OS for quite a long time (since Feb or so) and it's great. I'm using it on a Raspberry Pi B+, through WiFi and using a PiCam NoIR. I've installed this once, left the camera mounted on a wall and updated it through the UI since then. Now it has the latest available release (20150719).

I've always had, however, issues when playing movie files downloaded to my computer. I guess this issue is something weird or maybe a bad config, because otherwise it would already have been reported or would be in FAQ...

Normally I use VLC (on a Windows machine) and it just shows the first frame and stays like that. Other media players don't recognise the files at all. On an Android phone I can play them kind of fine (I don't have control over the video like FF or RW, but it does play the whole video). The videos are always saved as "*.avi.mpeg", and changing file extensions has made no difference at all. Normally I use 800x480 and 2 FPS, but I've also tried with higher resolutions without success.

I guess this issue has something to do with a non-clean install or maybe a movie format config not properly set up in my RPi. I think I should get the Pi, format the SD and reinstall MotionPie, but as the camera is not exactly at hand right now (even in another city), I wanted to know if anyone has an idea about what could be wrong or a way to fix it. If not, I will just do that and update this post with my results at the weekend.

Thanks a lot in advance!

donjakobo, 2015-08-31:
You might be describing a codec issue for the video.  *avi8 is just a _container_ and does not mean the video in avi is always playable.  In most cases the OS has to support the codec in the container for it to play it (hence windows gives you an error using standard media player).

jpl90, 2015-08-31:
Hi donjakobo,
thanks for your answer. As far as I know, VLC should take care of that and support the video codec automatically (as long as the file is correctly encoded, of course).
Is it possible that video encoding has some issue/misconfiguration or so?
Thanks again for your time

donjakobo, 2015-08-31:
@jpl90 yes... I stated VLC takes care of the codecs, but on Android or other you don't get the same (unless you are using VLC on Android as well, which I would find strange).

I personally have not checked what codec it is, but you can use VLC to view that info and determine support.  I wish motionEye would allow you the selection of *file type/codec*

jpl90, 2015-08-31:
Yes, I'm using VLC on Android as well (and QuickPic, too). I might be wrong, but I still think there is some issue on how MotionPie encodes the video.

I guess I will just have to make a clean install and see the results after that. I'll update soon.

Thanks!

yeahbert, 2015-09-05:
Had the same problem and fixed it by [changing the demuxer](http://superuser.com/questions/468880/vlc-2-frozen-stuck-on-first-frame-of-avi-video)

jpl90, 2015-09-13:
Perfect! Thank you @yeahbert, it worked.

Sorry for the delay, I took some time, got the board and made a clean install. Now, videos are not saved as "*.avi.mpeg" anymore and I feel the UI might be a little smoother.

Thanks!

closed 2015-09-13 by jpl90