ccrisan/motionpie #283: PiTFT monitor

rockman55, 2015-08-31:
Will MotionPie work on one of these small monitors (I am using an Adafruit PiTFT Plus Assembled 320x240 2.8" TFT + Resistive Touchscreen - Pi 2 and Model A+ / B+ PRODUCT ID: 2298) and the software is also from Adafruit 2015-02-16-raspbian-pitft28r_150312.
Thanks in advance.
Danny

ccrisan, 2015-08-31:
motionPie for RPi has only one display driver and that's the BCM2835/BCM2708 framebuffer. It won't work with any third party displays (i.e. those not connected to HDMI or RCA) and even if it worked, it would only display boot logs and a (apparenty useless) login prompt.

closed 2015-08-31 by ccrisan

rockman55, 2015-08-31:
Thanks for the input.
Danny

https://github.com/ccrisan/motionpie/issues/283#issuecomment-136282089