ccrisan/motionpie #284: Cannot check photos/videos: An error occurred. Refreshing is recommended.

donjakobo, 2015-08-31:
I keep getting this error today, yesterady all was fine, and I was able to pull up the web interface an check photos taken by motion capture.

Today I get *An error occurred. Refreshing is recommended.* in the UI
I've refreshed the view, also rebooted remotely, but still cannot access historical images.
Nothing was changed since yesterday and it was working fine for 2 days (setup is only 3 days old now).

In *motioneye.log* I see:
```
2015-08-31 12:27:57: ERROR: Exception in callback <functools.partial object at 0xb18a0d50>
Traceback (most recent call last):
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/ioloop.py", line 568, in _run_callback
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/stack_context.py", line 275, in null_wrapper
  File "/programs/motioneye/src/thumbnailer.py", line 71, in _run_process
    _process.start()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 130, in start
  File "/usr/lib/python2.7/multiprocessing/forking.py", line 121, in __init__
OSError: [Errno 12] Cannot allocate memory
```

donjakobo, 2015-08-31:
As an update, I rebooted multiple times, and now got to SSH into the rpi:

```
2015-08-31 15:41:16: ERROR: motion not running, starting it
2015-08-31 15:41:16: ERROR: failed to start motion: [Errno 12] Cannot allocate memory
Traceback (most recent call last):
  File "/programs/motioneye/motioneye.py", line 348, in checker
    motionctl.start()
  File "/programs/motioneye/src/motionctl.py", line 124, in start
    process = subprocess.Popen(args, stdout=log_file, stderr=log_file, close_fds=True, cwd=settings.CONF_PATH)
  File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
  File "/usr/lib/python2.7/subprocess.py", line 1231, in _execute_child
OSError: [Errno 12] Cannot allocate memory
```

So looks like motion can't start, but the webui runs.

oopeekoo, 2015-09-01:
Have to say this is continuous issue in my RPI2 setup. System works 2-3 days then something like this happens. Logs show that motion has run out of memory. Only recovey is to powercycle.
I don't even try to use MotionPie atm, just waiting a new version to test. If that fails, I have to move on to something else.

ccrisan, 2015-09-01:
Do you have many media files (i.e. pictures or movies)? A relatively large number of such media files will make motionEye eat up a lot of memory.

oopeekoo, 2015-09-01:
Me? Well, not really. The sd card was all the time relatively empty.

ccrisan, 2015-09-01:
Hmm, in that case there must be some configuration scenario that triggers this problem. Can you please send me your config files (/data/etc/motion.conf and /data/etc/thread-*.conf), *removing the sensitive data (email addresses, passwords, etc)*? Also I'd like to see all the [logs](https://github.com/ccrisan/motionpie/wiki/Reporting-An-Issue#attach-log-files).

ccrisan, 2015-09-01:
Another useful information would be, if you can login using SSH/putty, the output of the `ps aux` command at a moment when the error occurs, so that I can see who's eating up that much memory.

oopeekoo, 2015-09-01:
I have to re-enabe the device for this, might take some days to get errors in log.

ccrisan, 2015-09-01:
It might be worth identifying the bug.

donjakobo, 2015-09-01:
I'll look into grabbing the logs, I ended up "fixing" this issue by killing the motioneye.py process as it was just trying to start motion.py repeatedly and failing.  After killing both, cpu was down to 0, and I re-started it from the command line.

My SD was empty, had nothing to do with SD space.  I'll grab logs later and post.

donjakobo, 2015-09-02:
And again today the same issue:
```
2015-09-02 08:32:00: ERROR: Exception in callback <functools.partial object at 0xb607ad50>
Traceback (most recent call last):
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/ioloop.py", line 568, in _run_callback
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/stack_context.py", line 275, in null_wrapper
  File "/programs/motioneye/src/thumbnailer.py", line 71, in _run_process
    _process.start()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 130, in start
  File "/usr/lib/python2.7/multiprocessing/forking.py", line 121, in __init__
OSError: [Errno 12] Cannot allocate memory
```

No other error in the log leading up to it, just see 8/31 entries, and 9/02 errors start, device was NOT restarted. Everything is still running but CANNOT open viewing photo thumbnails for motion viewing.

I'm on a Raspberry 1 (512 ram, 2USB)

ccrisan, 2015-09-02:
And the output of `ps aux` in such a situation?

donjakobo, 2015-09-02:
@ccrisan here is the `ps aux`: http://pastebin.com/JscC5THt

oopeekoo, 2015-09-03:
I'm on day 2, no issues so far. Noticed that motioneye.log size is 11 MB, mostly because it is full of old "Cannot allocate memory..." errors. To clean up the log, you have to do it outside of the gui, yes?

ccrisan, 2015-09-03:
How about the [logs](https://github.com/ccrisan/motionpie/wiki/Reporting-An-Issue#attach-log-files)?

donjakobo, 2015-09-03:
I cleaned out my logs the other day due to them being at ~13MB of just repeated "out of memory" messages.  I'll run for 2 more days and see if I encounter this issue again.  I've also reduced the GPU Memory to 96mb from 128 before.  I'm guessing this might be a Python garbage collection issue? Or out of memory python message as the OS is still chugging along when I SSH in.

I'll post logs fully if/when I get this in a day or 2.

donjakobo, 2015-09-05:
Ok, got the out of memory again, attached all logs and zipped them up here:  http://wikisend.com/download/477980/MotionPi-OutofMemory-Logs.zip

Running only the default Rpi cam and using motion capture (photos, not video as was too choppy).

ccrisan, 2015-09-07:
It seems there's a serious memory problem with motionEye in certain configurations. I have yet to find the root cause of it.

oopeekoo, 2015-09-07:
Here's mine motioneye.log:
http://wikisend.com/download/679390/motioneye.log
Just noticed it has crashed. Took a while now, almost full week.

donjakobo, 2015-09-10:
Any detail on the logs that were posted? This seems like a python memory leak, as the OS doesn't run low.

ccrisan, 2015-09-12:
There is a memory leak problem reported by many users. I'm investigating the source of this problem and will come back with details.

jsmacdou, 2015-09-28:
Have you made any headway regarding this issue? A solution to bring the system back to life in the interim would be much appreciated as well. For me, the problem persists after rebooting.

ccrisan, 2015-09-29:
I prepared a special motionEye version that helps me investigating memory leaks. Unfortunately you cannot install it on motionPie; you need to install it on a normal OS (such as Raspbian). I'm still waiting for results from other users that have reported the problem.

mikecarr, 2015-10-14:
I thought I was having a non-motion detection issue until I looked at the logs and see I am having the same issues. I am not on RPI2, regular RP.

donjakobo, 2015-10-15:
Any closer on this update? motionPie is unusable for me as it just locks up 24+ hours into running.

mikecarr, 2015-10-15:
I am going to see when it craps out then set a cron to restart, should be
ok until this gets resolved.

donjakobo, 2015-10-15:
@mikecarr looks like @ccrisan actually quietly created a new fork of motionpie and called it motioneyeOS, I just installed the latest release: https://github.com/ccrisan/motioneyeos

mikecarr, 2015-10-15:
Let me know how it works.

ccrisan, 2015-10-15:
@donjakobo yes I did create the motionEyeOS project but it definitely wasn't "quielty". Check out the [main motionPie page](https://github.com/ccrisan/motionpie), the readme part at the bottom. I also posted it on [facebook](https://github.com/ccrisan/motionpie) as well as on the [Raspberry PI Forum](https://www.raspberrypi.org/forums/viewtopic.php?p=827765#p827765). I couldn't possibly comment on all issues or anything of that sort.