ccrisan/motionpie #285: USB cameras crashing

mattbot88, 2015-08-31:
Hi, I want to start out by saying I absolutely love motionpie. I have a problem, though. I am running 20150719 on a raspberry pi 2. I have 3 USB webcams, all Logitech. I wanted to put them in my house on vacation for fun. I set my router to port forwarding, so I can view the cameras and see when motion was detected anywhere, and it worked. I set them up, went to Starbucks to see if it would work outside of our network, and it came up with the live stream (just the live stream, no GUI. Just live stream.)I then added email notifications, which worked like a charm. Before I left for vacation, I plugged in my pi to see if it would overheat. (It got a little warm, but didn’t overheat)I also didn’t want the pi to get a new IP address so I tried not to reboot it too many times. Luckily it didn’t get a new IP address. Well, after I got to the hotel, I tried to access it and it didn’t work. When I got home I rebooted it and it worked on my own network. My webcams have a slider that can block the camera from seeing anything. I have those closed. When I try to access it, it works as supposed to. But when I open them, they all crash. Please reply. Thank you.
 											
Logs:
boot http://s000.tinyupload.com/?file_id=03951973712694582547
dmesg http://s000.tinyupload.com/?file_id=49543343511825833848
messages http://s000.tinyupload.com/?file_id=36993580795477716282
motion http://s000.tinyupload.com/?file_id=26349490509856767450
motioneye http://s000.tinyupload.com/?file_id=54277296302310913489
model numbers of webcams:
v-u0026 
v-ubk45
 v-u0007

ccrisan, 2015-09-01:
Hmm, the `boot.log` file has some lines that indicate errors on USB. It's hard to tell for sure what the cause to the problem is but I suggest the following:

1) make sure the cameras have enough power (use a powered HUB?)
2) if the cameras are different, try to test with only one type of camera and see which one missbehaves (it may be a driver problem)
3) if step (2) does not give a faulty camera model, then having all 3 at once on the USB of a PI may be the problem; PI's USBs are known to be problematic sometimes - you could use one camera per PI
4) as a last resort, if the problem still persists, you could trigger a systematic reboot at midnight using [crontab](https://github.com/ccrisan/motionpie/wiki/Tweaks#running-commands-at-regular-intervals-cron-support) and set a static IP to make sure you'll always have the same address.

mattbot88, 2015-09-01:
I will give it a try, thank you so much! :-þ

mattbot88, 2015-09-02:
Two out of three of the cameras work by themselves. the other goes on and off.

mattbot88, 2015-09-02:
tee hee sorry

closed 2015-09-02 by mattbot88