ccrisan/motionpie #291: Don't work with wi-fi

andrew81, 2015-09-07:
Hi! I try connect wi-fi and it doesn't work.
I do by instruction.
My wi-fi http://www.aliexpress.com/item/150Mbps-150M-Wireless-USB-WiFi-Wi-Fi-Wi-Fi-Adapter-With-External-Antenna-Wholesale-Free-Drop/1463345800.html 
and router Dlink DIR615.
Thanks

ccrisan, 2015-09-07:
Can you give me the output of the command `lsusb`, run in an SSH shell?

andrew81, 2015-09-07:
I haven't experience how do it. Pleas help.

Digital-Noob, 2015-09-10:
Hi, I got the same problem with this adapter. Here's the lsusb info that u need. Hope u can help solving this issue. Thank you.

Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp.
Bus 001 Device 008: ID 04fc:05d8 Sunplus Technology Co., Ltd Wireless keyboard/mouse
Bus 001 Device 005: ID 148f:7601 Ralink Technology, Corp.
Bus 001 Device 006: ID 1241:1603 Belkin Keyboard

andrew81, 2015-09-12:
Bus 001 Device 002: ID 0424:9512
Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 003: ID 0424:ec00
Bus 001 Device 004: ID 148f:7601
Bus 001 Device 005: ID 046d:08a2

andrew81, 2015-09-14:
Anybody can help?

ccrisan, 2015-09-14:
I will look into this and try to identify the missing driver. In the meantime, can someone tell me if this works with the latest raspbian?

Digital-Noob, 2015-09-14:
Hi, to answer to your question, it doesn't work with latest raspbian. I did try link this link for the solution, I managed to get the adapter detected & it able to scan the wireless network but I couldn't get it connected entering the passphrase, Push button & Enter PIN. 

https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=119095&p=809449

Let me elaborates some points. I did purchased 4pcs of the wireless wifi adapters with antenna off ebay from 2 different sellers. Both adapters look exactly the same. The 2nd purchase cost cheaper than the other one, almost half the price so I bought 2pcs. 

WORKING Adapter --- >Bus 001 Device 005: ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter
NOT WORKING Adapter  ---> Bus 001 Device 005: ID 148f:7601 Ralink Technology, Corp.

I have no problem with the 148f:5370.

Hope you able to help us to get this adapter working with motionpie & raspbian as I don't want to throw it away.

[image]

In the pic, both adapters have different packing. The left one is not working.

Thank you.

ccrisan, 2015-09-14:
If it has no support for raspbian yet, then the driver has not been mainlined yet in kernel 4.1. I won't include out-of-tree drivers in motionPie, because it's too difficult to maintain them across various boards and kernel versions. As soon as they will have support for Raspbian, motionPie will follow.

closed 2015-09-14 by ccrisan

Digital-Noob, 2015-09-14:
Hi @ccrisan , thanks for your prompt reply. I greatly appreciate your tremendous effort in making the motionpie a great tool for my home surveillance. I have 1 unit running for few months without any issue. 

Sorry to touch on this issue again, I know u already closed this thread at the moment. 

UPDATE: After I visit the link again https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=119095&p=809449 right after post it in this thread, I noticed I missed out this update

sudo wget https://github.com/porjo/mt7601/raw/master/src/mcu/bin/MT7601.bin -O /lib/firmware/mt7601u.bin

Finally with this update I managed to get the adapters to work. Hope this will also help others that encountered this issue, especially to @andrew81 who started this thread.

Again, thank you very much @ccrisan  : )

ccrisan, 2015-09-14:
If it's just a matter of adding a firmware file, I'll make sure to include it with the next release :)

Digital-Noob, 2015-09-14:
Thank you so much. I will be looking forward to that. Your effort is greatly appreciated : )

danyo41, 2015-12-18:
Hi there, I have read through the forum and everything seems to be exactly what I am experiencing. I know that was a while ago and things have probably changed. but I am currently using a Ralink MT7601U that works perfect right out of the box with raspbian but it will not work with MP. I have tried doing everything above and have tried doing an ifconfig but no wlan seems to come up, only eth0. Any help would be greatly appreciated. Sort of new to Linux but have been learning on my own for about a year now.

ccrisan, 2015-12-18:
@danyo41 have you switched over to [motionEyeOS](https://github.com/ccrisan/motioneyeos)?