ccrisan/motionpie #292: SSL certificate problem

bjumy, 2015-09-07:
Hi 

i have a raspberry pi 2 and i have hooked up an ip camera to it. 
The rasperry has motionpi and is connected to my network 192.168.30.171
Now the camera is on 30.199 but when i try to add the camera in the motionpi web interface
i get this error: SSL certificate problem self signed certificate.

I need some help i have been at this for ages

thanks

ccrisan, 2015-09-07:
Well, how about not using HTTPS? Does your camera support simple HTTP?

bjumy, 2015-09-08:
https://192.168.30.199/login.html?uri=/

this is the url im using i have tried multiple ways,

192.168.30.199

192.168.30.199:8080

https://192.168.30.199

https://192.168.30.199:8080

http://192.168.30.199

the only other error i got was not a supported network camera when i gave the login url of the web UI of my ip camera, but it's an ubuquti network camera.

ccrisan, 2015-09-08:
You need to use the MJPG HTTP URL of your camera. I honestly wouldn't know what that URL is, but you could look it up online, depending on your camera model. Have you tried http://192.168.30.199:8080?

bjumy, 2015-09-08:
it says connection refused with that link

bjumy, 2015-09-08:
rtsp://192.168.30.199:554/live/ch00_0

i also tried this one but it refuses the connection aswell

bjumy, 2015-09-08:
/live/ch00_0 is needed in the link i think

ccrisan, 2015-09-10:
I thought you solved it by adding the /live/ch... You still need to figure out the url yourself. I couldn't possibly know it.